polarofficial/polar-ble-sdk #262: iOS SDK not connecting

RubenTerra, 2022-04-08:
***Platform your question concerns:*** 
- [ ] Android 
- [1 ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ 1] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

Hello, 

I am trying to use the IOS SDK and I have run the iosBLETestApp. The app says that the watch has connected and provides information about the device (the name, id, etc) but all the watch displays is a loader that says "Connecting to device" -> "Connecting to Polar Flow App". The device then throws an error: "Couldn't connect to the Polar Flow App". 

Would you be able to help? 
Thank you!

JOikarinen, 2022-04-10:
Hi @RubenTerra,

thanks for your question. From your question I get the impression that you are trying to connect iosBLETestApp with one of the Polar wrist units, right? Polar BLE SDK works only with [Polar sensors
](https://github.com/polarofficial/polar-ble-sdk#sdk-for-polar-sensors), like Polar Verity Sense.

RubenTerra, 2022-04-10:
Hello @JOikarinen, 

Thank you for your response. 

Is there any way of connecting an iOS app with a polar wrist unit? 

Thank you, 
Ruben

JOikarinen, 2022-04-11:
Yes, Polar wrist units can be connected and synced with the [Polar FlowApp](https://apps.apple.com/fi/app/polar-flow/id717172678?l=en). Polar FlowApp is the app for Polar wrist unit users to analyze they daily activities recorded by wrist units and sync data to [Polar Flow](https://flow.polar.com/) cloud. But you question probably concerns whether there is a application which can provide data for 3rd party developers over bluetooth straight from wrist unit, for that question the answer is no at the moment.  

However, I would like to highlight there is of course an alternative approach, depending on what use cases you are interested in. Polar provides API for accessing cloud data with https://www.polar.com/en/accesslink, please have a look. Also if you are familiar with some other application which may provide 3rd party support of your interest you may take a look on this site https://www.polar.com/en/compatible-apps