polarofficial/polar-ble-sdk #263: Recording-related APIs are not working on H10

cuihujun, 2022-04-10:
Platform your question concerns:
- [x] Android

Device:
- [x] Polar H10

Description:

these recording-related APIs are not working on H10, when using `api.requestRecordingStatus` to check the recording status, there is no ongoing recording.

- api.startRecording
- api.stopRecording

Logs
`
2022-04-10 10:55:13.763 10354-10381/com.m.mypolar D/PolarDataSourceImpl: ECG: ts=702903312619511941 samples.size=73
2022-04-10 10:55:13.943 10354-10354/com.m.mypolar D/PolarDataSourceImpl$callback: hrNotificationReceived: 91
2022-04-10 10:55:14.321 10354-10354/com.m.mypolar D/ViewRootImpl@43b92e2[MainActivity]: ViewPostIme pointer 0
2022-04-10 10:55:14.335 10354-10381/com.m.mypolar D/PolarDataSourceImpl: ECG: ts=702903313179873286 samples.size=73
2022-04-10 10:55:14.409 10354-10354/com.m.mypolar D/ViewRootImpl@43b92e2[MainActivity]: ViewPostIme pointer 1
2022-04-10 10:55:14.411 10354-10354/com.m.mypolar I/PolarDataSourceImpl: startRecording: 
2022-04-10 10:55:14.442 10354-10354/com.m.mypolar W/com.m.mypolar: Accessing hidden field Ljava/nio/Buffer;->address:J (light greylist, reflection)
2022-04-10 10:55:14.573 10354-10438/com.m.mypolar E/PolarDataSourceImpl: com.polar.androidcommunications.api.ble.model.gatt.client.psftp.BlePsFtpUtils$PftpResponseError: Request failed:  Error: 209
        at com.polar.androidcommunications.api.ble.model.gatt.client.psftp.BlePsFtpClient.readResponse(BlePsFtpClient.java:643)
        at com.polar.androidcommunications.api.ble.model.gatt.client.psftp.BlePsFtpClient.lambda$query$5$com-polar-androidcommunications-api-ble-model-gatt-client-psftp-BlePsFtpClient(BlePsFtpClient.java:442)
        at com.polar.androidcommunications.api.ble.model.gatt.client.psftp.BlePsFtpClient$$ExternalSyntheticLambda4.subscribe(Unknown Source:6)
        at io.reactivex.rxjava3.internal.operators.single.SingleCreate.subscribeActual(SingleCreate.java:40)
        at io.reactivex.rxjava3.core.Single.subscribe(Single.java:4813)
        at io.reactivex.rxjava3.internal.operators.single.SingleSubscribeOn$SubscribeOnObserver.run(SingleSubscribeOn.java:89)
        at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run(Scheduler.java:614)
        at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:65)
        at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:56)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:764)
2022-04-10 10:55:14.961 10354-10381/com.m.mypolar D/PolarDataSourceImpl: ECG: ts=702903313740260061 samples.size=73
2022-04-10 10:55:14.996 10354-10354/com.m.mypolar D/PolarDataSourceImpl$callback: hrNotificationReceived: 91
2022-04-10 10:55:15.450 10354-10381/com.m.mypolar D/PolarDataSourceImpl: ECG: ts=702903314300616318 samples.size=73
`

JOikarinen, 2022-04-11:
Hi @cuihujun, 

from the logs I can see that you have received `Error: 209`. That means the H10 battery is too low for recording start.

JOikarinen, 2022-04-11:
Made a fix in https://github.com/polarofficial/polar-ble-sdk/commit/3d4b561e362eb324d39a777507f65d8281e0dff6 to map the magic error codes to human readable format.

closed 2022-04-11 by JOikarinen