polarofficial/polar-ble-sdk #265: ECG using Polar H10 and Arduino

StiNor27, 2022-05-03:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I want to read the ECG values that my Polar H10 is reading by transfering it to an Arduino RP2040 nano. To do so I have asked the Arduino to read the service 180D and characteristic 2a37. When I do this I get a string of data that I am having trouble to interpret. 
Here is the output I am receiving:

11:28:56.968 -> ----------------------------
11:28:56.968 ->          SenSe 
11:28:56.968 ->   Polar H10 pulsmonitor
11:28:56.968 -> ----------------------------
11:28:57.063 -> Found 06:12:95:8f:ae:da '' 
11:28:57.253 -> Found f8:a6:6f:94:aa:27 'Polar H10 9EF6E824' 180d
11:28:57.253 -> Connecting ...
11:28:57.301 -> Connected
11:28:57.301 -> Discovering service 180D ...
11:28:57.682 -> Service discovered
11:28:57.682 -> Subscribing to simple key characteristic ...
11:28:57.778 -> Subscribed
11:28:58.160 -> 16 88 25 3 15 3 0 
11:28:59.161 -> 16 86 1 3 15 3 0 
11:29:00.165 -> 16 85 219 2 179 2 0 
11:29:01.168 -> 16 85 160 2 179 2 0 
11:29:02.173 -> 16 87 147 2 139 2 0 
11:29:03.176 -> 16 88 142 2 139 2 0 
11:29:04.176 -> 16 89 151 2 127 2 0 
11:29:05.167 -> 16 90 122 2 127 2 0 
11:29:06.171 -> 16 92 133 2 155 2 0 
11:29:07.174 -> 16 92 185 2 155 2 0 
11:29:08.175 -> 16 91 192 2 224 2 0 
11:29:09.179 -> 16 90 242 2 224 2 0 
11:29:10.183 -> 16 88 191 2 164 2 0 
11:29:11.187 -> 16 88 139 2 164 2 0 
11:29:12.142 -> 16 90 144 2 165 2 0 
11:29:13.144 -> 16 90 191 2 165 2 0 
11:29:14.149 -> 16 89 210 2 198 2 0 
11:29:15.152 -> 16 89 193 2 198 2 0 
11:29:16.204 -> 16 89 168 2 144 2 0 
11:29:17.204 -> 16 89 145 2 144 2 0 
11:29:18.160 -> 16 90 147 2 161 2 0 
11:29:19.162 -> 16 91 171 2 161 2 0 
11:29:20.164 -> 16 91 164 2 171 2 0 
11:29:21.167 -> 16 91 143 2 127 2 0 
11:29:22.167 -> 16 91 122 2 127 2 0 
11:29:23.167 -> 16 93 125 2 134 2 0 
11:29:24.170 -> 16 93 153 2 134 2 0 
11:29:25.176 -> 16 93 138 2 118 2 0 
11:29:26.180 -> 16 94 113 2 118 2 0 
11:29:27.185 -> 16 95 126 2 167 2 0 
11:29:28.233 -> 16 95 216 2 167 2 0 
11:29:29.236 -> 16 92 226 2 198 2 0 
11:29:31.243 -> 16 91 157 2 198 2 0 
11:29:31.243 -> 16 91 135 2 133 2 0 
11:29:32.200 -> 16 92 149 2 133 2 0 
11:29:33.203 -> 16 92 178 2 207 2 0 
11:29:34.209 -> 16 91 197 2 207 2 0 
11:29:35.211 -> 16 90 171 2 148 2 0 

As far as I understand the pulse is the second bit, and the rr-interval is the third? Is this right, and how do I get the ECG data? Do I need to use other characteristics and services?

JOikarinen, 2022-05-05:
Hi @StiNor27,

> As far as I understand the pulse is the second bit, and the rr-interval is the third?
- 0x180D is the Heart Rate service defined by the BluetoothSig. You may have a look on specification in [Heart Rate Service](https://www.bluetooth.com/specifications/specs/heart-rate-service-1-0/) and [Heart Rate Profile](https://www.bluetooth.com/specifications/specs/heart-rate-profile-1-0/). The standard Heart Service is specifying the how to interpret the data received from Heart Rate service. You may have a look how the SDK library HR client is interpreting the data received from the HR service: https://github.com/polarofficial/polar-ble-sdk/blob/610d5e1a4cb38d05d873794055e60a44616bcbe5/sources/Android/android-communications/library/src/main/java/com/polar/androidcommunications/api/ble/model/gatt/client/BleHrClient.java#L68

> how do I get the ECG data? Do I need to use other characteristics and services?
- 0x180D is not specifying method to deliver ECG data over BLE, so that's why Polar H10 has proprietary BLE service for the purpose. That is service is documented here https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf

Gerken80, 2022-06-06:
How are you going with this? I have just started similar with Nano33BLE. I can read and understand the 180d and battery services no problem no problem. Now just started working on the other features, EGR etc.

apar945, 2023-06-15:
> * 0x180D is not specifying method to deliver ECG data over BLE, so that's why Polar H10 has proprietary BLE service for the purpose. That is service is documented here https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf

Hello 

I can't find the above mentioned document. Seems like it has the information for anyone using their own bluetooth application to collect ECG data.

Please suggest if this information is available elsewhere.

Ankit

andYgnite, 2023-07-06:
Hello,

document was removed from repository after SDK release 5.0.
You can still find document in older commits:
https://github.com/polarofficial/polar-ble-sdk/tree/39c809ab4a3398d3ea88ff8a0cc6f6d44cd58bda/technical_documentation

closed 2023-09-08 by jimmyzumthurm

pascalzellner, 2025-01-30:
I ll try to find the ECG characteristic GUID number for an H10 and I found this post here the method to understand each octets packet 

Using the peak-peak method, the firmware measures each RR (wave R to wave R) interval. Using the RR value, it calculates the heart rate (HR) = 60/RR in seconds. Read the [heart service specifications](https://www.bluetooth.com/specifications/specs/heart-rate-service-1-0/) to analyse each packet [XX, XX, XX, XX].
The first octet is the FLAGS number. Each bit is a specific flag :
Bit 0: Heart Rate Value Format (0 = UINT8, 1 = UINT16).
Bit 2: Sensor Contact Status (0 = Not Supported, 1 = Supported).
Bit 1: Sensor Contact Status (0 = Not Detected, 1 = Detected).
Bit 3: Energy Expended Status (0 = Not Present, 1 = Present).
Bit 4: RR-Interval (0 = Not Present, 1 = Present).
Bits 5–7: Reserved for future use.
After the first octet, we constantly analyse the packet in the same order :
Heart Rate Value
If Flag - Bit 0 is 0, the heart rate value is in the next byte (UINT8, range 0–255 BPM).
If Flag - Bit 0 is 1, the heart rate value is in the next two bytes (UINT16, range 0–65535 BPM).

Sensor Contact Status
If Flag - Bit 1 is 1, Flag indicates the sensor contact status - Bit 2:
Flag - Bit 2 = 0: No contact detected.
Flag - Bit 2 = 1: Contact detected.

Energy Expended
If Flag - Bit 3 is 1, the next two bytes represent the energy expended (UINT16, in kilojoules).
RR-Interval
If Flag-Bit 4 is 1, the subsequent bytes represent RR intervals (UINT16, in 1/1024 seconds).
For example, we receive from the firmware server [22, 58, 248, 1, 252, 0, 252, 0, 251, 0]. The first octet decimal value (sometimes you will have hexadecimal values) is 22, 0X16 in hexadecimal and 00010110 in binary :
00010110, bit 0 (the rightest bit) is zero, so the heart rate value is UINT8 using one octet.
00010110, bit 2 is one; the sensor supports the skin contact status.
00010110, bit 1 is one; the sensor detects the skin contact for this packet.
00010110, bit 3 is zero, and the energy expended status is not present, so for this packet, all octets after the 8 UINT heart rate octet will be octets to calculate UINT16 RR values if bit 4 is one.
00010110, bit 4 is one; the subsequent bytes represent RR-Intervals.

The FLAGS packet is a dynamic one. It can vary for each measure. Each Flag guides us in making available calculations. Because bit 0 signs a UINT8 value, we jump to the  next octet to find the heart rate of 58 (or 3A in hexadecimal) bpm. Bit 3 is zero, so all the subsequent packets will be RR octet :
Remember RR are UINT16; we take the two subsequent values, 248 and 1, to obtain  the RR value: 248 is 0xF8 in hexadecimal, and 1 is 0x01. This formula gives RR value `(0x01 << 8) | 0xF8` = `0x01F8` = `504` in decimal. But 504 is a fact, the number of 504 units of 1/1024 seconds. We generally use RR in milliseconds, and the final result is (504/1024)*1000, in fact, 492.19 ms.
This method uses the next double octets to obtain the following RR values: 246.09 ms, 246.09 ms, and 245.12 ms.