polarofficial/polar-ble-sdk #266: Unable to connect on Samsung Galaxy Tab A8 running Android 11

lschmierer, 2022-05-04:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
We are unable to connect to the Polar H10 on Samsung Galaxy Tab A8 running Android 11.
After initial connection we get a timeout:
`D/BluetoothGatt: onConnectionUpdated() - Device=E5:C4:70:DD:2F:E7 interval=6 latency=0 timeout=500 status=0`

It is worth noting that the official Polar Beat app does not seem to work as well.

**How to Reproduce**
Run the provided example app and try to connect to Polar H10.

Alternatively try the official Polar Beat app.

**Expected behavior**
Being able to connect and persist the connection to the device. 

**Screenshots and logs**
```
I/BluetoothAdapter: enable()
D/InputMethodManager: HSIFW - flag : 0
D/InputMethodManager: prepareNavigationBarInfo() DecorView@4a97318[MainActivity]
D/InputMethodManager: getNavigationBarColor() -855310
V/InputMethodManager: Starting input: tba=com.example.MyApp ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
D/InputMethodManager: startInputInner - Id : 0
I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
W/IInputConnectionWrapper: requestCursorAnchorInfo on inactive InputConnection
D/InputTransport: Input channel constructed: 'ClientS', fd=170
D/InputTransport: Input channel destroyed: 'ClientS', fd=162
D/InsetsSourceConsumer: setRequestedVisible: visible=false, type=13, host=com.example.MyAppApp/com.example.MyApp.MainActivity, from=android.view.InsetsSourceConsumer.hide:233 android.view.ImeInsetsSourceConsumer.hide:101 android.view.InsetsController.hideDirectly:1444 android.view.InsetsController.controlAnimationUnchecked:1120 android.view.InsetsController.applyAnimation:1431 android.view.InsetsController.hide:992 android.view.ViewRootImpl$ViewRootHandler.handleMessage:6259 android.os.Handler.dispatchMessage:106 android.os.Looper.loop:247 android.app.ActivityThread.main:8676 
D/InputMethodManager: HSIFW - flag : 0
I/BluetoothAdapter: onBluetoothStateChange: up=true
I/BluetoothAdapter: onBluetoothStateChange: Bluetooth is on
D/InsetsAnimationCtrlImpl: finish: control=android.view.InsetsAnimationControlImpl@a5ed396, types=8, mCancelled=false, mCancelRequested=false
I/SurfaceControl: nativeRelease nativeObject s[-5476376624668223856]
I/SurfaceControl: nativeRelease nativeObject e[-5476376624668223856]
I/BluetoothAdapter: STATE_ON
I/BluetoothAdapter: STATE_ON
I/BluetoothAdapter: STATE_ON
I/BluetoothAdapter: STATE_ON
D/BluetoothLeScanner: Start Scan with callback
D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=9 mScannerId=0
I/BluetoothAdapter: STATE_ON
I/BluetoothAdapter: STATE_ON
D/BluetoothLeScanner: Stop Scan with callback
D/MyApp: CONNECTING: 84615125
I/BluetoothAdapter: STATE_ON
I/BluetoothAdapter: STATE_ON
D/BluetoothGatt: connect() - device: E5:C4:70:DD:2F:E7, auto: false
I/BluetoothAdapter: isSecureModeEnabled
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=7e9ddc2c-6307-4ba7-bd27-9c8d840f1880
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=E5:C4:70:DD:2F:E7
D/MyApp: CONNECTED: 84615125
D/BluetoothGatt: discoverServices() - device: E5:C4:70:DD:2F:E7
D/BluetoothGatt: onConnectionUpdated() - Device=E5:C4:70:DD:2F:E7 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt: onClientConnectionState() - status=62 clientIf=9 device=E5:C4:70:DD:2F:E7
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=9
I/BluetoothAdapter: STATE_ON
I/BluetoothAdapter: STATE_ON
I/BluetoothAdapter: STATE_ON
I/BluetoothAdapter: STATE_ON
D/BluetoothLeScanner: Start Scan with callback
D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=9 mScannerId=0
D/MyApp: DISCONNECTED: 84615125
I/BluetoothAdapter: STATE_ON
I/BluetoothAdapter: STATE_ON
D/BluetoothLeScanner: Stop Scan with callback
```

JOikarinen, 2022-05-05:
Hi @lschmierer,

that is something we have worked hard lately. At the moment the investigations of the root cause has lead us all the way down to the specific chipset manufacturer which is used on few of the latest phone and tablets. That same problematic chipset is used on [Samsung Galaxy Tab A8](https://www.gsmarena.com/samsung_galaxy_tab_a8_10_5_(2021)-11265.php) too.  We still continue the investigations whether we could enhance the Polar side to make connectivity to work.

lschmierer, 2022-05-05:
Hi @JOikarinen,

thanks for your quick response!
Just to clarify, is this an issue with the chipset not behaving in a spec-conformant way or with the Polar devices?
If the former, it would theoretically be the manufacturers obligation to fix this issue. Unfortunately this would probably take months to work its way through the Android updates process, if possible at all. So a workaround on your side is the only realistic way to address the issue, as far as I see it.

The question is, however, important to us, because if this is an issue with the the chipset not behaving in a spec-conformant way, we might be able to return the tablets we purchased under warranty law (as they are faulted). We were planning on using the devices in a medical study and already bought a bunch of tablets and H10 trackers.

JOikarinen, 2022-05-06:
> Just to clarify, is this an issue with the chipset not behaving in a spec-conformant way or with the Polar devices?
I guess what you have experienced is that you cannot connect with Polar or other BLE devices if using the Polar BLE SDK. It seems that problem to create the connection can be solved by changes on Polar BLE SDK. I will try to finish and publish those changes in coming days here in GitHub. 

However, still there are problems even the connection is created successfully. There are moments the communication between the phone and device just stops. At the moment the analysis is that we cannot fix the problem by changes either on  Polar devices or on this SDK.

lschmierer, 2022-05-06:
We will probably switch to another Android device in the short-term.

Is there any compatibility list so that we do not run into issues again?
(will e.g. [Galaxy Tab A7 Lite](https://www.gsmarena.com/samsung_galaxy_tab_a7_lite-10933.php) or [Lenovo Tab M8](https://www.gsmarena.com/lenovo_tab_m8_(hd)-9825.php) work?)

JOikarinen, 2022-05-07:
I believe, both of those tablets you mentioned shall work fluently.  👍

UjjwalGupta, 2022-05-15:
I am facing same issue in Samsung Galaxy Tab A8, Android 11. Trying to connect Wahoo ticker. There is inconsistency in device connection. Sometimes it connect, sometimes it taking long time but not connecting. 
If anyone can find better solution, please leave a comment.
Thank you.

JOikarinen, 2022-06-03:
There is now enhancement on this issue published in [Release 3.3.2](https://github.com/polarofficial/polar-ble-sdk/releases/tag/3.3.2). However the full workaround haven't been found yet. The status remains that some PolarBleSDK features are not working if phone/tablet has specific model of Unisoc chipset.

What is not working:
- the data streams are not working, i.e. any function related to streaming. For example like [startEcgStreaming](https://github.com/polarofficial/polar-ble-sdk/blob/6a967212b116fc4cd4edc567026d3206b12f7ff1/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L396)

What is working
- BLE connections are working
- HR data is working
- H10 offline recording is working

vietnt-uet, 2022-07-27:
Note sure it is still an issue, we are working on Galaxy Tab A8 and we can stream PPG, GYR, and ACC well.

jimmyzumthurm, 2023-09-08:
I will close this as it has been stale for over a year. If you still have issues or questions related to that, please feel free to re-open.

closed 2023-09-08 by jimmyzumthurm