polarofficial/polar-ble-sdk #270: Initialization issue of PolarBleApiDefaultImpl

kcbaruri, 2022-05-17:
***Platform your question concerns:*** 
- [ ] Android 

***Device:***
- [ ] Other: Wahoo Tickr

**Description:**
In an application, we are trying to initialize the ble api as follows -
mPolarBleApi = PolarBleApiDefaultImpl.defaultImplementation(this, PolarBleApi.FEATURE_HR);

But an exception is being raised as below
com.polar.sdk.api.errors.PolarBleSdkInstanceException: Attempt to create Polar BLE API with features 1. Instance with features 255 already exists

Are we missing anything or the error is being occurred due to non-polar device?

kcbaruri, 2022-05-17:
In the older version of the SDK, the code was working accordingly but in the latest one, we are experiencing the above issue.

JOikarinen, 2022-05-18:
Hi @kcbaruri, that indicates that the Polar SDK is instantiated twice. Somewhere there is another instantiation with the option ```PolarBleApiDefaultImpl.defaultImplementation(this, PolarBleApi.ALL_FEATURES);```.

If possible please refactor your app so that Polar SDK is instantiated only once. Alternatively you may instantiate SDK multiple times, but before you instantiate another SDK instance you must call [shutDown](https://github.com/polarofficial/polar-ble-sdk/blob/938b689f8a88eba0049f1786e96d133441e69271/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L140) for previous instance.

closed 2022-05-18 by JOikarinen

kcbaruri, 2022-05-18:
Hi JOikarinen,
Yes, you are right and yesterday we noticed that and resolved it.
Thanks