polarofficial/polar-ble-sdk #271: Unresolved reference: PolarBleApi

kitkatbonsai, 2022-05-23:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [x] Device is not relevant for this question

**Description:**
I've followed the installation guide for Android in https://github.com/polarofficial/polar-ble-sdk but I cannot get this code work for me in the main activity when making a new activity from scratch, it looks that it could not import properly the dependencies: 

private val api: PolarBleApi by lazy {
        // Notice PolarBleApi.ALL_FEATURES are enabled
        PolarBleApiDefaultImpl.defaultImplementation(applicationContext, PolarBleApi.ALL_FEATURES)
    }

I am using Android Studio ChipMunk and gradle. distributionUrl=https\://services.gradle.org/distributions/gradle-7.3.3-bin.zip

I have also had to replace the line:
repositoriesMode.set(RepositoriesMode.FAIL_ON_PROJECT_REPOS)
with
repositoriesMode.set(RepositoriesMode.PREFER_SETTINGS)

to make this code work in gradle.build:
buildscript {
    ext.kotlin_version = '1.6.21'
    repositories {
        mavenCentral()
        google()
    }

    dependencies {
        classpath 'com.android.tools.build:gradle:7.2.0'
        classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:$kotlin_version"
    }
}
allprojects {
    repositories {
        mavenCentral()
        google()
        maven { url 'https://jitpack.io' }
    }
}

Whereas in gradle.build.app I have included the required repos within the dependency tree:

dependencies {
...
    implementation 'io.reactivex.rxjava3:rxjava:3.1.3'
    implementation 'io.reactivex.rxjava3:rxandroid:3.0.0'
    implementation 'com.github.polarofficial:polar-ble-sdk:3.3.1'
}

Ultimately, it all "works" but I get these warnings on build finish:
Failed to resolve: https://jitpack.io
Failed to resolve: com.github.polarofficial:polar-ble-sdk:3.3.1

Formerly I used to import the polar sdk with and external *.aar in my own libs directory, but I wanted to know what am I doing wrong here, and if there is a way to import the polar sdk as any other dependency with gradle. I also checked the example in kotlin. I have used exactly the same gradle files and I cannot get why it does not work when doing another porject from scratch. Thank you in advance for the response!

kitkatbonsai, 2022-05-23:
At the end of the day it was a likely inexact information issue on the installation guide. Specifically, the installation guide says on step 2: 

```
2. Add the JitPack repository to your build file
allprojects {
    repositories {
        ...
        maven { url 'https://jitpack.io' }
    }
}
```

But it does not specify in which "build file", there are two of them anmd these are not referenced to the code presented! (build.gradle - Project and build.gradle - Module *.app).

For those starting with Android (like me), keep in mind that the maven-related code has to be placed in settings.gradle file, and not in any *.buld file:

```
pluginManagement {
    repositories {
        gradlePluginPortal()
        google()
        mavenCentral()
    }
}
dependencyResolutionManagement {
    repositoriesMode.set(RepositoriesMode.PREFER_SETTINGS)
    repositories {
        google()
        mavenCentral()
        **maven { url "https://jitpack.io" }**
    }
}
rootProject.name = "MainActivity"
include ':app'
```

This way, for a brand new activity or Android Studio project, you don't have to change anything. At the end, the build.gradle(Project) is left "as it was":

```
// Top-level build file where you can add configuration options common to all sub-projects/modules.
plugins {
    id 'com.android.application' version '7.2.0' apply false
    id 'com.android.library' version '7.2.0' apply false
    id 'org.jetbrains.kotlin.android' version '1.6.21' apply false
}

task clean(type: Delete) {
    delete rootProject.buildDir
} 
```

The only thing with this newflanged jetpack.io-based method is that instead of downloading and importing an *.aar file (as we had to do in the past in order to use the Polar API) we can just include it as a conventional dependency in the build.gradle(Module *.app) file, as said in the installation guide (https://github.com/polarofficial/polar-ble-sdk) and in the official jetpack.io documentation (https://jitpack.io/docs/#:~:text=FAQ-,JitPack.io,project%20build%20and%20upload%20steps.), e.g.:

```
dependencies {
    ...
    implementation 'io.reactivex.rxjava3:rxjava:3.1.3'
    implementation 'io.reactivex.rxjava3:rxandroid:3.0.0'
    implementation 'com.github.polarofficial:polar-ble-sdk:3.3.1'
}
```

**Conclusion**:
The Step 2 in the installation guide presented in README.md file within https://github.com/polarofficial/polar-ble-sdk can be confusing. The line of code _maven { url "https://jitpack.io" }_ should be specified to be included in settings.gradle within the dependencyResolutionManagement block, as specified above.

I don't really know why it worked this way... if there is a kind person that can shed some light to this, I would be very thankful.

Happy coding!

JOikarinen, 2022-05-24:
Hi @kitkatbonsai,

thank you for the good explanations and observations. First I was about to say that `settings.gradle` is not the recommended location to place repository settings. But the [Google document](https://developer.android.com/studio/build) exactly supports your suggestion how the Android gradle project shall be setup nowadays. After little search I found that starting from Android Gradle Plugin Version 7.1 the recommended way is the use of [centralized repository declarations](https://docs.gradle.org/current/userguide/declaring_repositories.html#sub:centralized-repository-declaration) in `settings.gradle`. 

At the moment [example project](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp) is still following the "old" style gradle project setup and it is working. However, as you explained, if the new app project is started in Android Studio the template gradle setup follows the new style and the README is not up to date. I will reopen this issue and will fix the README, also setup the example project to follow the centralized repository declarations.

kitkatbonsai, 2022-05-24:
Thank you very much @JOikarinen!

closed 2022-05-25 by JOikarinen