polarofficial/polar-ble-sdk #274: Get live data on Pc

Chromwurst, 2022-05-28:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

Has anyone experience in which is the best way to get the live data directly on to your pc? Unfortunately I have no experience with Bluetooth programming yet.

JOikarinen, 2022-05-30:
Hi @Chromwurst, unfortunately standard answer on this GitHub page is this:

Unfortunately, we are only able to support Android and iOS. The SDK includes source code and you can feel free to utilize / port the code in other platforms too but that is something we are not able to provide support for.

closed 2022-05-30 by JOikarinen

JOikarinen, 2022-05-30:
@Chromwurst there are projects in GitHub where other platforms than Android and iOS are supported. For example: https://github.com/cjs30/FingerPulseLatency