polarofficial/polar-ble-sdk #277: Problems with new version Polar SDK

abrusdev, 2022-06-01:
**Platform on which you observed the bug:** 
- [*] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [*] Device is not relevant for this bug

**Describe the bug**
When device is not connected to Polar Device and at the same time, if i try to use methods of  instance of (PolarBleApiDefaultImpl), like shutDown(), foregroundEntered(), backgroundEntered(), SDK is crashing. Older SDK was worked fine.

**How to Reproduce**
Just get instance of PolarBleApiDefaultImpl, but don't connect to any device. Then try to call methods of this instance, like foregroundEntered(), shutDown().

**Expected behavior**
As before, if the method can't be done, just ignore it, but don't throw NullPointerException.

**Screenshots and logs**
[image]
[image]

abrusdev, 2022-06-01:
https://github.com/polarofficial/polar-ble-sdk/compare/master...abrusdev:patch-1

JOikarinen, 2022-06-02:
Hi @abrusdev, 

thanks for the good description. I am trying your steps with [example app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp). The example app does similar steps what you describe, when app is started the instance of PolarBleApi is created by calling ` PolarBleApiDefaultImpl.defaultImplementation` and later when switching app background and then foreground the api `foregroundEntered()` is called without a problem. 

But if you call the [`shutDown()`](https://github.com/polarofficial/polar-ble-sdk/blob/8f52ccf514de36cf5b5d6258b46f0a0f06754208/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L140) you shall initialise the `PolarBleApi` again by calling the ` PolarBleApiDefaultImpl.defaultImplementation`. That is because `shutDown` releases all the resources kept by the `PolarBleApi` object.  The change how the `shutDown()` releases the resources was made in release [3.2.3](https://github.com/polarofficial/polar-ble-sdk/releases/tag/3.2.3). 

So this is what I expect happens in your app is that the `shutDown()` is called at some point, but later there are still calls to `PolarBleApi`.

You are correct that if method cannot done it shall just silently ignore, I will take a look.

abrusdev, 2022-06-02:
Hi, @JOikarinen. I will also recheck my code. Thank you for your comment.

closed 2022-06-02 by abrusdev

JOikarinen, 2022-06-03:
@abrusdev this issue is now fixed in [release 3.3.2](https://github.com/polarofficial/polar-ble-sdk/releases/tag/3.3.2)