polarofficial/polar-ble-sdk #279: Renaming the device

maneetgoyal, 2022-06-10:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

Is it possible to rename the device using this SDK?

For example, my device name is Polar Sense XXXXXXXX. Can I rename it to "My Device" via SDK so that next time I try to connect to the device it shows me "My Device" in the scanner window?

JOikarinen, 2022-06-13:
Hi @maneetgoyal, unfortunately that's not possible. What kind of use case you have, probably we could try to figure out alternative solution?

maneetgoyal, 2022-06-13:
Thanks for the reply @JOikarinen.

>What kind of use case you have, probably we could try to figure out alternative solution?

I am building a web application that connects to multiple Polar Verity Sense devices, subscribes to their heart rate and battery services, and displays them in a chart.

The use case is to simply rename the device and make the new name persist on the device itself so that any PC/Mobile/etc. can identify the device via its new name in the future. `Polar Sense XXXXXXXX` is hard to remember when you have multiple sensors deployed.

Would love to study any alternates you may propose. I considered persisting the new name in the web app instead ([`localStorage`](https://developer.mozilla.org/en-US/docs/Web/API/Window/localStorage)) but then the problem is that **only** that particular instance of the app will know the new name, not the other instances. Alternately. the new name can also be stored in a cloud database so that it can be fetched on-demand. But I am restricted in that the app is supposed to work in offline mode.

maneetgoyal, 2022-06-13:
Also, can you please let me know if the heart rate (in BPM) can be obtained at a reduced sampling frequency? Currently, once I trigger [`startNotficiations`](https://developer.mozilla.org/en-US/docs/Web/API/BluetoothRemoteGATTCharacteristic/startNotifications) I get a new value each second. I was wondering if can get this value at say, every 200ms?

JOikarinen, 2022-06-13:
> Also, can you please let me know if the heart rate (in BPM) can be obtained at a reduced sampling frequency? Currently, once I trigger [...]

Let me start from this one. BLE heart rate service sends the heart rate data around once per second, that is specified in [BLE Heart rate service specification](https://www.bluetooth.com/specifications/specs/heart-rate-service-1-0/), you may check the chapter "3.1.1.5 Transmission Interval".

JOikarinen, 2022-06-13:
> The use case is to simply rename the device and make the new name persist on the device itself so that any PC/Mobile/etc. can identify the device via its new name in the future. `Polar Sense XXXXXXXX` is hard to remember when you have multiple sensors deployed.

I understand. The "XXXXXXXX" is the deviceID and it is good identifier, but I suppose you have human factor involved? The human can't remember the deviceID when switching from, let say, PC to mobile. By quick thinking I cannot see other options than "hiding" the deviceID from user, let the user rename the device and the keep the map deviceId:user_given_name on you application🤷🏻‍♂️

maneetgoyal, 2022-06-13:
>The "XXXXXXXX" is the deviceID and it is good identifier, but I suppose you have human factor involved?

Yes, you're right it's a good identifier. And correct, there are human factors making it difficult to use.

>let the user rename the device and the keep the map deviceId:user_given_name

Feeling the same :)

jimmyzumthurm, 2023-09-08:
There is currently no plan to allow changing the device name for Verity Sense, although we understand that this might be an interesting feature for some uses cases.

closed 2023-09-08 by jimmyzumthurm