polarofficial/polar-ble-sdk #280: PPG STREAM DECODE ISSUE

huguessK, 2022-06-13:
***Platform your question concerns:*** 
- Platform is not relevant for this question

***Device:***
- Polar Verity Sense

**Description:**

I'm working with the Polar Verity Sense and I'm particularly interested in getting ppg data from the sensor through BLE communication using the Bleak library on Python.
However, I face a problem in the decoding of the ppg stream I receive from the sensor.

After writting to PMD control point [01 01]->0x01=Get_measurement settings, 0x01=measurement_type(PPG) , 
I got the following response from the Polar Verity Sense:
f0, 01, 01, 00, 00, 00, 01, 87, 00, 01, 01, 16, 00, 04, 01, 04

Following the technical_documentation provided on your github, I decode the response:
0xf0= control point response
0x01=op_code(Get Measurement settings)
0x01=meausrement_type(PPG)
0x00=error_code(success)
0x00=more_frames(false)
0x00=setting_type(SAMPLE_RATE), 0x01=array_length(1),0x87 0x00=135Hz
0x01=setting_type(RESOLUTION), 0x01=array_length(1), 0x16 0x00=22 bits
0x04=setting_type(CHANNELS)
0x01=array_length(1)
0x04=4 channels (ppg0,ppg1,ppg2,ambient0)

I then write to PMD control point the following bytes array [02, 01, 00, 01, 87, 00, 01, 01, 16, 00, 04, 01, 04]
for ppg stream request. I receive streams but don't know how to decode them.
Here is some of the streams I receive:

a ppg stream:

01, 8b, 41, 10, 1d, 90, d7, d4, 09, 80, 49, b8, fe, a9, c7, fe, b2, db,
fe, 98, e4, fb, 14, 14, 68, e8, a0, c7, 0c, 02, d5, a0, 12, 0e, 73, fa,
ff, 12, fd, 98, db, ef, f9, fe, 2c, 2f, df, 95, f1, b1, 1e, af, 7c, f2,
75, 14, df, cb, f2, 51, 26, ff, 99, f1, 76, 1c, 10, 15, 05, d6, 42, 50,
bf, 02, c9, 14, 21, bd, 11, 6a, 1a, 61, 76, 11, e4, ad, 20, ae, 08, eb,
94, a0, 54, 0a, 34, b2, 2f, dd, f8, 6c, 96, bf, 81, fa, b5, 09, 0f, fb,
ef, c5, 02, 6f, 5f, f0, 50, 45, 1f, 9a, f6, a5, 5f, 4f, d5, f4, c5, 7d,
d0, c7, 0a, c1, 9f, b0, c0, 08, 55, 17, 11, ed, 10, 43, 12, 71, 5a, 11,
5d, 6a, 00, ee, 03, 0a, 4a, 30, f5, 05, ce, 76, 3f, a7, f5, 90, 60, 7f,
f5, f6, a6, fa, 6e, ef, ef, f3, fd, fe, 9f, ef, 8f, 91, ef, 1b, fc, 4a,
b5, 9f, e6, f9, 1f, cf, b0, 13, 0f, 20, e8, 00, 98, 0d, 22, fe, 20, 76,
0e, 04, ef, a0, 82, 0f, 4f, 1d, b0, 1f, ff, d8, fb, 6f, 25, 01, 09, 46,
ef, c7, f2, 15, 34, 6f, ed, f3

another one:

01, d6, 16, e9, 33, 90, d7, d4, 09, 80, c5, f5, fd, 1c, bd, fd, 62, e5,
fd, 2e, 0d, fb, 14, 14, 84, ee, 0f, 2d, 02, ba, 13, 90, d1, ff, 25, 06,
71, 8c, 11, 96, 14, 11, c7, 10, 2b, ce, f0, d1, 0a, 68, b6, a0, 56, 0c,
27, d4, 2f, c8, fa, b4, b4, 4f, 99, fc, 33, 19, 0f, 7f, f0, 6d, 0d, 8f,
40, f1, 96, 2b, ef, ad, f4, bc, 42, bf, 1f, f3, b6, 52, e0, 4f, 08, 2a,
77, f0, 1b, 06, 3a, 1b, 51, ae, 11, 4f, 1c, 71, bc, 11, 43, 8c, e0, 49,
06, 01, 6f, f0, 24, 08, 2a, 90, 5f, 06, f7, 4d, 77, cf, 94, f8, d3, fc,
fe, 99, ef, 68, fb, ae, 95, ef, b1, 6a, ef, 74, f9, 13, 8b, 1f, 55, f7,
f9, ac, 90, 4d, 0d, 2e, ca, 90, 91, 0b, cb, 0d, 11, c2, 0f, 69, 02, e1,
8a, 10, 9a, 3f, 60, 3e, 01, 75, 1e, 30, 41, 03, 41, 5b, 5f, fe, f3, 0d,
47, bf, 3f, f5, eb, fc, ae, 8a, f0, f1, 05, ef, e5, ef, 6d, c3, af, 69,
ff, 13, e8, 7f, 18, fd, fd, f0, 80, aa, 10, ff, 03, 61, 99, 0f, e0, e3,
f0, 7f, 0c, 2a, d0, 40, c4, 0d

On the last stream :
0x01 = PPG 
[d6 16 e9 33 90 d7 d4 09] = last sample timestamp in nanoseconds
However I don't know what bytes represent ppg0, ppg1, ppg2, ambient0 and how to convert them to decimal values as it is not specified in the technical_documentation.

Coud you help me to decode the stream?
Thank you!

JOikarinen, 2022-06-15:
Hi @huguessK, the received data is in compressed format. The compression is briefly explained in chapter **Delta frame sample example from Polar Verity Sense (Acc data as example)** of [Polar_Measurement_Data_Specification](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf). I wonder is that enough information you can deserialize the PPG data?

huguessK, 2022-06-15:
Hi @JOikarinen,
 looking at the Acc data as example, I see that it is possible to get sample counts but in my case i'm unable to have it making it difficult to deserialize the PPG data.
Assuming, ppg data to start after byte 0x80 and ppg0, ppg1, ppg2, ambient0 on 3 bytes (as specified in the docmentation) each I have 12bytes per sample. 
The problem I face is that I get 224 bytes per PPG stream, as the bytes 1 = measurement type (PPG) ; bytes 2 to 9 = timestamp ; bytes 10 = Delta frame; I finally have 224-10=114 bytes which is not a multiple of 12

giulianosrebuli, 2024-02-28:
Hi @huguessK, don't know if it will be useful to you now, but the byte that has the 0x80 indicates that your sample is delta compressed, so you should expect the next 12 bytes to be the reference value (ppg0, 1, 2 and ambient) and then the number of bits per ppg sample. I had a similar problem but with acc data, as said in #443. Hope it helps

thebabyswiss, 2025-01-31:
I am also working on getting PPG data from my verity sense to stream out and save to my laptop via python. I used a code example to do so with ACC data and it works fine. When I try it with PPG, it does not work. Also, the byte array that it returns when I request the settings is:
\xf0\x01\x01\x00\x00\x00\x017\x00\x01\x01\x16\x00\x04\x01\x04

which is different from the documentation and all other examples I have found from other users. 
taking this output and using
02 01 00 017 01 01 16 00 04 01 04 as the input does not seem to work. 

Does anyone have a fix for  @huguessK would you mind providing your python code?

closed 2025-07-03 by rkangast