polarofficial/polar-ble-sdk #281: Streaming data with Pytho,n

Edouard99, 2022-06-13:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

Edouard99, 2022-06-14:
My bad I finally suceeded to activate notification and it is working properly

closed 2022-06-14 by Edouard99

JanMenssen, 2022-07-04:
Dear Edoard

I have the same problem, I want to read ECG data from the H10 using BLE using the arduinoBLE.
Reading the UID FB005C81-02E7-F387-1CAD-8ACD2D8DF0C8 results also in 0x0F 0x05 0x00

Writing to 0x01, 0x00 to the same UUID seems succesful, however no update status and reading again results in the same message and not in a control point response starting with 0xF0 0x01 0x00

So I guess, I should also activate the notification as mentioned in your post, however I do not know how to do this ?
So please, let me know how you solved it

regards - jan

JanMenssen, 2022-07-08:
problem solved ;-)

dara1400, 2023-07-14:
> problem solved ;-)

Hello

Could you explain how did you manage to solve the problem?

dara1400, 2023-07-14:
> My bad I finally suceeded to activate notification and it is working properly

Nice,
Could you explain how?

ccwang65, 2025-05-09:
I want to know if it is possible to use SDKMode to read ECG at maximum sampling rate. Appreciate if any one want to share.