polarofficial/polar-ble-sdk #284: Can the SDK return why it can't connect?

abrusdev, 2022-06-24:
***Platform your question concerns:*** 
- [*] Android 
- [*] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [*] Device is not relevant for this question

**Description:**
Good day. We were struggling with one problem. When the polar device is connected to two devices, the third device cannot connect to the polar device. This is a normal case, but it is somehow incomprehensible to the user. Are there any way to show the user a warning like "Your device is already connected to 2 devices". Can your SDK handle this kind of situation?

Thanks for your answer in advance.

JOikarinen, 2022-07-25:
Hi @abrusdev, 

thanks for the good question. Let's start with information related to your question:
 
- to enable 2 BLE connections with H10 or VeritySense the option need to be enabled in Polar FlowApp ([H10](https://support.polar.com/uk-en/support/can_i_pair_h10_with_several_devices) instructions / [VeritySense](https://support.polar.com/ie-en/can-i-use-verity-sense-with-several-devices) instructions). Default option is one BLE connection. 
- if Polar device cannot create more connections (it already has either 1 or 2 connections) it may either stop advertising itself or it continues advertising itself as non-connectable. If Polar device has ended advertising then`searchForDevice()` function ([android](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L365) / [ios](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApi.swift#L163)) cannot find the device. If the Polar device is still advertising but it is not connectable the device is still found by `searchForDevice()`. However, then you shall double check the field `isConnectable` in `PolarDeviceInfo` ([android](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarDeviceInfo.java#L32) / [ios](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApi.swift#L54))

And then more precise answer to your question:
- you may try to create a connection to Polar device using its deviceId (or BLE address) with `connectToDevice()`([android](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L268)/[ios](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApi.swift#L150)). 
    - if the device is not advertising then `connectToDevice()` function doesn't see the device at all and in this scenario SDK cannot help on distinction the reason for not connecting to device.
    -  if the Polar device is advertising but is in non-connectable state then `connectToDevice()` function shall be able to tell the reason why the connection is not possible. At the moment SDK only logs the error in this case. Logs like these are printed:
        Android:
        ```
        D/API LOGGER: ConnectionHandler/Skipped connection attempt due to reason device is not in connectable advertisement or missing service
        ```
        iOS:
        ```
       Polar SDK log:  [BLE] connection attempt deferred due to reason device is non connectable
       ```
        - I will investigate whether  `connectToDevice()` could throw an exception instead of just logging the error reason. 
 
Sorry for the late answer. 
-Jukka

JOikarinen, 2022-07-25:
Hi @abrusdev, 

> * I will investigate whether  `connectToDevice()` could throw an exception instead of just logging the error reason.

This is not easy to achieve, causes reasonable big changes `connectToDevice()` API. I wonder if the previous commented helped on your case? You can gain partial solution by using the `searchForDevice()` before using the `connectToDevice()` function.

abrusdev, 2022-08-02:
Thanks for your answer. We will solve this problem for now with the help of the isConnectable function.

closed 2022-08-02 by abrusdev