polarofficial/polar-ble-sdk #285: Arduino Library

zarpli, 2022-06-27:
hi,

I will publish in a few days the library for the Arduino universe, only for Verity Sense.

You will be able to obtain all the data that the sensors provide, and then use any favorite algorithm.

It is quite useful to be able to activate the VS (Verity Sense) without pressing its button when it is powered through its USB adapter, but it would also be very useful to be able to access BMP, PPI at the same time as PPG.

I hope that you will consider being able to use all the features that the sensor offers in SDK mode,

Ale.

zarpli, 2022-06-27:
[image]

jimmyzumthurm, 2023-09-08:
Hi @zarpli ,

First, great job about successfully porting the Polar SDK to Arduino ! 

> 
I hope that you will consider being able to use all the features that the sensor offers in SDK mode

Unfortunately the SDK mode has been designed with enabling custom settings for each sensor in mind, to enable applications that might need for example high accelerometer sampling rate for special motion detection algorithms.

HR and PPI are produced by our own proprietary algorithms, and these algorithms were validated with specific sensor settings. For example it needs PPG at 55 Hz and ACC at 52 Hz with -+ 8G range, so we cannot run these algorithm in SDK mode and expect them to produce reliable data.

With normal uses cases (non-SDK mode), we are not considering it normal use case that device would calculate HR when connected to USB cable, even if it is convenient for your use case. I hope it is enough if raw data stream are only available with USB cable connected when device is in SDK mode, even if I understand that you won't be able to get HR / PPI.

zarpli, 2023-09-08:
hi @jimmyzumthurm,

It is good to know the details and reasons for the SDK mode restrictions.

Due to some details with firmware 2.1.0, I did not release the library for Arduino.

I am working on some examples that will be included with the library, once finished and doing some tests with some colleagues, I will publish them,

thanks for your attention :)