polarofficial/polar-ble-sdk #286: Connection timeout / Connection attemp

kitkatbonsai, 2022-07-01:
***Platform your question concerns:*** 
- [ x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ x] Device is not relevant for this question

**Description:**

Hello there! First of all, thank you for the great work you're doing with our questions. This time, I was wondering if there was any way (or workaround) to check if a device ID is connectable or not (e.g., the device you want to connect is not found or it is turned off). Is there any connection timeout or something? Otherwise, the code is doing "nothing" when we try to connect a device that appears to be turned off. If the logic is "do nothing until you manage to connect to your device", this is ceretainly a blocking move. 

I know it may sound dumb, "connect to a device that is turned off", but in many contexts this is totally contemplated. Imagine you make a mistake by introducing the device ID or something. At least, there should be a function in the API to check so!

Thank you in advance for your answer!

JOikarinen, 2022-07-25:
Hi @kitkatbonsai,

thanks for the question. Your questions is a bit similar than #284 

I would suggest the same workaround than in #284, i.e. you may check the surrounding devices with `searchForDevice()` ([android](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L365) / [ios](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApi.swift#L163)) to be sure the device of interest is around and the device is connectable. Before calling `connectToDevice()`

> If the logic is "do nothing until you manage to connect to your device", this is ceretainly a blocking move.

Yes, this is true. `connectToDevice()`([android](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L268)/[ios](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApi.swift#L150)) tries to create a connection as long as `disconnectFromDevice` ([android](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L276) / [ios](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApi.swift#L156)) is called. 

Note, `setAutomaticReconnection` ([android](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L202) / [ios](https://github.com/polarofficial/polar-ble-sdk/blob/43e972d58638b3bae2e26e4940abde73545943d0/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApi.swift#L379)) controls whether `connectToDevice()` continues the connection attempt in the case already connected device connection is lost. 

Sorry for delayed response,
-Jukka

kitkatbonsai, 2022-07-25:
Alright, thank you very much!

JOikarinen, 2022-07-25:
@kitkatbonsai please reopen for more questions if needed.

closed 2022-07-25 by JOikarinen