polarofficial/polar-ble-sdk #287: Receiving data from the PMD Control point

JanMenssen, 2022-07-04:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

See https://github.com/polarofficial/polar-ble-sdk/issues/281

I have the same problem, I want to read ECG data from the H10 using BLE using the arduinoBLE.
Reading the UID FB005C81-02E7-F387-1CAD-8ACD2D8DF0C8 results also in 0x0F 0x05 0x00

Writing to 0x01, 0x00 to the same UUID seems succesful, however no update status and reading again results in the same message and not in a control point response starting with 0xF0 0x01 0x00

So I guess, I should also activate the notification as mentioned in the mentioned post, however I do not know how to do this ?
So please, let me know how you solved it

regards - jan

JanMenssen, 2022-07-08:
> _**Platform your question concerns:**_
>
> * [ ]  Android
> * [ ]  iOS
> * [ ]  Other
> * [ ]  Platform is not relevant for this question
>
> _**Device:**_
>
> * [ ]  Polar OH1
> * [ ]  Polar Verity Sense
> * [ ] [...]

Problem solved ;-)

AndreasLamparter, 2022-07-16:
Hi, i have the same problem, what is your solution ?
Best regards Andreas

AndreasLamparter, 2022-07-17:
I forgot to set the ATT MTU min requested size of 232.

closed 2022-07-25 by JOikarinen