polarofficial/polar-ble-sdk #288: How to implement setLocalTime

BeaconBrigade, 2022-07-07:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

I'm working on a rust library for the H10 [here](https://github.com/RoughSketch/arctic). I noticed that the timestamps for ecg and acc data were off. I've seen that you fix this while using the SDK with `setLocalTime`. My question is, how would you implement this?  Is there a uuid to write to, or do you write to the control point? Thanks

BeaconBrigade, 2022-07-15:
Hey @JOikarinen 

Is there any info related to this you can link? Thanks

JOikarinen, 2022-07-25:
Hi @BeaconBrigade,

sorry for long delayed answer. This is not nice, but unfortunately I must say this statement for non Android/iOS projects:

We are only able to support Android and iOS. The SDK includes source code and you can feel free to utilize / port the code in other platforms too but that is something we are not able to provide support for.

closed 2022-12-13 by JOikarinen