polarofficial/polar-ble-sdk #292: AppStore review issues

borisgolovnev, 2022-08-14:
***Platform your question concerns:*** 
- [ ] Android 
- [*] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [*] Device is not relevant for this question

**Description:**

I am trying to publish an app that shows ECG signal from Polar H10 sensor on the AppStore but Apple insists that by using Polar SDK I am infringing upon the rights of Polar Devices and that I need documentary evidence of authorisation to use the SDK. I tried sending them the SDK license but apparently a text file on github is not a valid form of authorisation. Guess they want an agreement or something. 

Any ideas on what can I do about this?

borisgolovnev, 2022-08-14:
Never mind, they changed their mind. It only took 7 re-submissions each with increased befuddlement un my part.

closed 2022-08-14 by borisgolovnev

JOikarinen, 2022-08-15:
@borisgolovnev good the issue was finally resolved.👌

nikashitsa, 2024-07-30:
I faced exactly the same issue. @borisgolovnev what did you tell Apple Review team?

borisgolovnev, 2024-07-30:
Nothing specific that worked, just kept replying that the SDK is open source and available to everyone and re-submitting the app over and over and eventually they let it through. I don't think AppStore reviewers can be reasoned with, pretty sure most of them don't even read or try the app because they don't care. And in this case, given that the app requires specific hardware to work, I don't think they really can test it so when they see that it was rejected before they will just reject it again and go on with their day. So you just need go get a reviewer that doesn't care and will let the app through instead of rejecting it and for that you just need to try a bunch of times.

nikashitsa, 2024-07-31:
@borisgolovnev thanks for answer. I've tried only 4 times. Going to continue submitting 😊

nikashitsa, 2024-07-31:
5th attempt was successful. My app is now in AppStore 🎉  
I wrote in "App Review Information" following Notes: 
```
This application is using public Polar SDK (https://www.polar.com/en/developers) which is open source and available to everyone under the License https://github.com/polarofficial/polar-ble-sdk/blob/master/Polar_SDK_License.txt

From text of License above:
"Polar is willing to grant You a limited license to use the Polar Software Development Kit software to develop, test and commercialize (if You so select) Your Licensee Software on the terms and conditions set forth herein."

Please have a look at attached demo.
```

And attached demo video.