polarofficial/polar-ble-sdk #295: ERROR_INVALID_MTU error while enabling sdkMode

Vietvantue, 2022-08-24:
**Platform on which you observed the bug:** 
- [ x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
We connect Polar Verity Sense to our sample app in Android run on Galaxy Tab A8 SM-X200 and try to enable SDK mode.  It gets **ERROR_INVALID_ MTU**
Our Polar Verity Sense firmware version is 1.1.5

**How to Reproduce**
1. Run sample android app
2. Connect to Polar Verity Sense device
3. Enable SDK Mode

**Expected behavior**
Can enable SDK Mode normally

**Screenshots and logs**
[image]

JOikarinen, 2022-08-24:
Hi @Vietvantue, do you have another Android device (i.e. phone or tablet) which you could use to test your sample application? Galaxy Tab A8 SM-X200 may fall into to "problematic devices" category. Galaxy Tab A8 SM-X200 has the Unisoc chipset, which has caused some problems with BLE. See the [issue 266](https://github.com/polarofficial/polar-ble-sdk/issues/266).

Vietvantue, 2022-08-24:
@JOikarinen We use another Tab A8 SM-X205 with Unisoc chipset and it works well. As far as I know, they are both using Unisoc Tiger T618 (12 nm) chipset.

JOikarinen, 2022-08-29:
Hi @Vietvantue, sorry for delayed reply. 

The root cause for whole problem is this. The Unisoc chipset has the problem that once the chipset negotiates the BLE MTU parameters with the Polar device (or some other device) there is cases that negotiated MTU parameters are not respected while data transfer. Because of that communication easily brakes between Polar device and Unisoc chipset. At the moment I don't have working solution on the problem. 

However, your case is a quite of interesting, as you told the  Tab A8 SM-X205 and Tab A8 SM-X200 are pretty identical tablets. If Tab A8 SM-X205 is working well, may you double check the customer has the latest updates in Tab A8 SM-X200? That's first quick check comes in to my mind.

Vietvantue, 2022-08-29:
@JOikarinen Thank you for your information. Our customer decided to procure a new tablet [Samsung Galaxy Tab S7 FE](https://www.gsmarena.com/samsung_galaxy_tab_s7_fe-10922.php) with a Snapdragon chipset. Do you think we are okay with this new device?

JOikarinen, 2022-08-30:
@Vietvantue Samsung Galaxy Tab S7 FE shall work fine.

closed 2022-08-30 by Vietvantue

Vietvantue, 2022-10-12:
@JOikarinen I need one more help from you, beside Tab A8 (Android 11), do we have any other devices that have issues with connecting to Polar sensors? And is it correct that the Intel chipset devices will have no problem with Polar sensors?
Thank you in advance.

jimmyzumthurm, 2023-09-08:
Hi @Vietvantue , you can find here a list of known problematic devices : https://github.com/polarofficial/polar-ble-sdk/blob/fdb5bd0c485eecbfc857708f8f9c65efd02052ae/sources/Android/android-communications/library/src/main/java/com/polar/androidcommunications/common/ble/PhoneUtils.kt#L14