polarofficial/polar-ble-sdk #297: exercise stuck on the h10

smyffanon, 2022-09-04:
Hello

I am using the android example, and am recording my sleeping RR-intervals - readings are from 5 to 8 hours long. 

A couple of times a month a reading will not be readable from the h10. 

Listing the  works fine, and it seems the app begins reading, but does not finish, maybe there is an error in the format on the 10 - maybe a division by zero somewhere? 

If I erase the exercise, then the next night it usually works fine again. 

questions

1. is there another way to pull the readings off the h10, so I can see if there's something wrong in the RR interval data
2. will removing the battery erase the exercise? or could it be that removing the battery might fix the issue but the exercise is still there? (I doubt it) 
3. how do I set up android studio to debug this? 

I was a programmer a long time ago but android development is pretty new to me. 
Although since this works 29 out of 30 days this doesn't look like an Android issue, seems like an h10 thing.

smyffanon, 2022-09-04:
forgot to include - I"ll leave the exercise on there for a couple of days, hoping I can debug the issue.

While it's on there, polar beat and polar sensor logger won't read the HR
but Polar Sensor Logger seems to be getting the ECG fine.

JOikarinen, 2022-09-05:
Hi @smyffanon,
thanks for the question. I understand your question and the challenges you have faced. 

> 1. is there another way to pull the readings off the h10, so I can see if there's something wrong in the RR interval data
- there can be 3rd party developed apps which might do the reading of RR file from the H10 memory, but unfortunately I am not familiar with any of them. From Polar side there is the example app which you have been using already, and Polar Beat. However Polar Beat can read only exercises it has started by it self. 

> 2\. will removing the battery erase the exercise? or could it be that removing the battery might fix the issue but the exercise is still there? (I doubt it)
- battery removal won't erase the exercise. 

> 3\. how do I set up android studio to debug this? I can connect android studio to the app on the phone, but an exception doesn't get caught as a breakpoint, so I don't know on what line of code it's throwing.
- do you see on the Android Studio logcat screen what is the exception example is throwing? You might be right that RR file is somehow corrupted in H10 thus reading it causes exception thrown by the example application. Another possibility that something happens during the transfer of the big exercise (i.e. over the night recording in your case) from H10.

smyffanon, 2022-09-05:
Thanks

Is  there a way to keep my h10 turned on and transmitting while it is off my body? 

It has been transmitting now over an hour and I have no idea when it will finish. 

There are sometimes 29,000 samples in my overnight recording, and the logcat is processing 2 per second

JOikarinen, 2022-09-21:
Hi @smyffanon, sorry for long delay with respond. 

> Is there a way to keep my h10 turned on and transmitting while it is off my body?
- by this question you mean this scenario:

1. You have done overnight offline recording, the data is saved to H10 memory
2. While reading out the recording from H10 memory you wear off the H10 -> and your question is will H10 keep on while reading out, even though the read out may take long

smyffanon, 2022-10-03:
Sorry for my own long delay.  I was sick for an extended period & just looked at this agaiin. 

This is an instruction on how to discharge an h10 after the battery is removed - I assume there must be some capacitors on there and they need a bit of discharging. 

The instruction is to moisten 2 fingers and kept the fingers touching the contact tabs.

I was thinking of placing some wet cotton on the two contact points and weighting these down with a bit of wood. 

The issues would be
 1. this would not make a completed circuit, which the 2 wet fingers technique does, so this may not work if a complete circuit is needed
 2. are there any issues of damaging my h10 if I keep it in this state for a long time artificially?

JOikarinen, 2022-10-06:
Hi @smyffanon ,

The release [3.3.4](https://github.com/polarofficial/polar-ble-sdk/releases/tag/3.3.4) speeds up the internal recording reading significantly. I hope that will slightly help you on this.

closed 2022-10-06 by JOikarinen