polarofficial/polar-ble-sdk #300: Several rxjava3 exceptions (OnErrorNotImplementedException and UndeliverableException)

biosignalssolutions, 2022-09-22:
**Platform on which you observed the bug:** 
- [X ] Android 

**Device on which you observed the bug:**
- [X ] Polar H10

**Describe the bug**
The  last 28 days there are an icrease of crahses (17 events) in reports of Google play console

**How to Reproduce**
I don't know how reproduce

**Expected behavior**
Maybe sometines the connection is lost

**Screenshots and logs**
CRASH : CASE 1

io.reactivex.rxjava3.exceptions.OnErrorNotImplementedException
Exception io.reactivex.rxjava3.exceptions.OnErrorNotImplementedException: The exception was not handled due to missing onError handler in the subscribe() method call. Further reading: https://github.com/ReactiveX/RxJava/wiki/Error-Handling | java.lang.NullPointerException
  at io.reactivex.rxjava3.internal.observers.EmptyCompletableObserver.onError (EmptyCompletableObserver.java:50)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableSubscribeOn$SubscribeOnObserver.onError (CompletableSubscribeOn.java:74)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableFromAction.subscribeActual (CompletableFromAction.java:40)
  at io.reactivex.rxjava3.core.Completable.subscribe (Completable.java:2859)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableSubscribeOn$SubscribeOnObserver.run (CompletableSubscribeOn.java:64)
  at io.reactivex.rxjava3.android.schedulers.HandlerScheduler$ScheduledRunnable.run (HandlerScheduler.java:123)
  at android.os.Handler.handleCallback (Handler.java:938)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at android.os.Looper.loop (Looper.java:237)
  at android.app.ActivityThread.main (ActivityThread.java:8090)
  at java.lang.reflect.Method.invoke
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:656)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:967)
Caused by java.lang.NullPointerException:
  at com.polar.sdk.impl.BDBleApiImpl.deviceStateMonitorObserver$lambda-63$lambda-61 (BDBleApiImpl.kt:868)
  at com.polar.sdk.impl.BDBleApiImpl.$r8$lambda$gDQgw_Ntp7iiyDa21L7nkc7mIzA
  at com.polar.sdk.impl.BDBleApiImpl$$ExternalSyntheticLambda67.run
  at io.reactivex.rxjava3.internal.operators.completable.CompletableFromAction.subscribeActual (CompletableFromAction.java:36)
  
 ----------------------
  CRASH : CASE 2
  io.reactivex.rxjava3.exceptions.UndeliverableException
Exception io.reactivex.rxjava3.exceptions.UndeliverableException:
  at io.reactivex.rxjava3.plugins.RxJavaPlugins.onError (RxJavaPlugins.java:372)
  at io.reactivex.rxjava3.internal.util.AtomicThrowable.tryAddThrowableOrReport (AtomicThrowable.java:52)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.innerError (FlowableFlatMap.java:564)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableFlatMap$InnerSubscriber.onError (FlowableFlatMap.java:636)
  at io.reactivex.rxjava3.internal.operators.single.SingleToFlowable$SingleToFlowableObserver.onError (SingleToFlowable.java:68)
  at io.reactivex.rxjava3.internal.operators.single.SingleDelayWithCompletable$OtherObserver.onError (SingleDelayWithCompletable.java:64)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableAndThenCompletable$SourceObserver.onError (CompletableAndThenCompletable.java:62)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableSubscribeOn$SubscribeOnObserver.onError (CompletableSubscribeOn.java:74)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableCreate$Emitter.tryOnError (CompletableCreate.java:91)
  at com.polar.androidcommunications.api.ble.model.gatt.BleGattBase.lambda$waitNotificationEnabled$1$com-polar-androidcommunications-api-ble-model-gatt-BleGattBase (BleGattBase.java:417)
  at com.polar.androidcommunications.api.ble.model.gatt.BleGattBase$$ExternalSyntheticLambda1.subscribe
  at io.reactivex.rxjava3.internal.operators.completable.CompletableCreate.subscribeActual (CompletableCreate.java:40)
  at io.reactivex.rxjava3.core.Completable.subscribe (Completable.java:2859)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableSubscribeOn$SubscribeOnObserver.run (CompletableSubscribeOn.java:64)
  at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run (Scheduler.java:644)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run (ScheduledRunnable.java:65)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call (ScheduledRunnable.java:56)
  at java.util.concurrent.FutureTask.run (FutureTask.java:266)
  at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run (ScheduledThreadPoolExecutor.java:301)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at java.lang.Thread.run (Thread.java:923)
Caused by com.polar.androidcommunications.api.ble.exceptions.BleDisconnected:
  at com.polar.androidcommunications.api.ble.model.gatt.BleGattBase.lambda$waitNotificationEnabled$1$com-polar-androidcommunications-api-ble-model-gatt-BleGattBase (BleGattBase.java:404)

JOikarinen, 2022-09-23:
Thanks for the report @biosignalssolutions,  let's figure out fix for this. The version of Polar BLE SDK is [3.3.2](https://github.com/polarofficial/polar-ble-sdk/releases/tag/3.3.2) or older? Is the crash happening only with certain Android version (like Android 13?) or Android version don't matter?

JOikarinen, 2022-09-23:
> Just to double check the CRASH : CASE 1 and CRASH : CASE 2 are independent? i.e. you see in the crashes in your crash analytics tools and the crashes are not related, those happen at different time on different users?

I figured out the CASE 1 and CASE 2 are separate issues. CASE 1 shall be now fixed in release [3.3.3](https://github.com/polarofficial/polar-ble-sdk/releases/tag/3.3.3). I was able to reproduce the CASE 2, but I need some time to figure out the root cause and proper fix.

biosignalssolutions, 2022-09-23:
Hello,

Thanks for your quick replay. 

The version of Polar BLE SDK is 3.3.2
I think Android version it doesn't matter. 
The CRASH : CASE 1 and CRASH : CASE 2 are independent, and it happen at different time on different users.

I'm not sure that origin of the problem is from the Polar SDK. I attach some image statistics, but the images contain data of other crashes.
It would be great to know **how/where to catch** these exceptions, so at least the app would continue to work.

CRASH CASE 1: (6 users affected)
Android 11 (SDK 30). 15 crash events (83%)
Android 12 (SDK 31), 3 crash events (16%)

CRASH CASE 2: (14 users affected)
Android 12 (SDK 31), 13 crash events (52%)
Android 11 (SDK 30), 12 crash events (48%)

Just another crash
CRASH CASE 3: (7 users affected)
Android 11 (SDK 30), 5 crash events
Android 12 (SDK 31), 2 crash events 

Type
io.reactivex.rxjava3.exceptions.UndeliverableException
Exception io.reactivex.rxjava3.exceptions.UndeliverableException:
  at io.reactivex.rxjava3.plugins.RxJavaPlugins.onError (RxJavaPlugins.java:372)
  at io.reactivex.rxjava3.internal.util.AtomicThrowable.tryAddThrowableOrReport (AtomicThrowable.java:52)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.innerError (FlowableFlatMap.java:564)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableFlatMap$InnerSubscriber.onError (FlowableFlatMap.java:636)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableDoOnEach$DoOnEachSubscriber.onError (FlowableDoOnEach.java:111)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableSubscribeOn$SubscribeOnSubscriber.onError (FlowableSubscribeOn.java:102)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableDoFinally$DoFinallySubscriber.onError (FlowableDoFinally.java:90)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableCreate$BaseEmitter.errorDownstream (FlowableCreate.java:299)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableCreate$BufferAsyncEmitter.drain (FlowableCreate.java:539)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableCreate$BufferAsyncEmitter.signalError (FlowableCreate.java:489)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableCreate$BaseEmitter.tryOnError (FlowableCreate.java:287)
  at com.polar.androidcommunications.common.ble.RxUtils.postError (RxUtils.java:45)
  at com.polar.androidcommunications.common.ble.RxUtils.postDisconnectedAndClearList (RxUtils.java:25)
  at com.polar.androidcommunications.api.ble.model.gatt.client.BleDisClient.reset (BleDisClient.java:61)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.BDDeviceSessionImpl.handleDisconnection (BDDeviceSessionImpl.java:312)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.BDDeviceListenerImpl$4.deviceDisconnected (BDDeviceListenerImpl.java:469)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.connection.ConnectionHandler.deviceDisconnected (ConnectionHandler.kt:122)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.GattCallback.onConnectionStateChange$lambda-2 (GattCallback.kt:46)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.GattCallback.$r8$lambda$IG57x4NQXNPQUMf9eA1FlMk5iLg
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.GattCallback$$ExternalSyntheticLambda3.run
  at io.reactivex.rxjava3.internal.operators.completable.CompletableFromAction.subscribeActual (CompletableFromAction.java:36)
  at io.reactivex.rxjava3.core.Completable.subscribe (Completable.java:2859)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableSubscribeOn$SubscribeOnObserver.run (CompletableSubscribeOn.java:64)
  at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run (Scheduler.java:644)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run (ScheduledRunnable.java:65)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call (ScheduledRunnable.java:56)
  at java.util.concurrent.FutureTask.run (FutureTask.java:266)
  at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run (ScheduledThreadPoolExecutor.java:301)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at java.lang.Thread.run (Thread.java:923)
  
smilar crash
CRASH CASE 4:
  io.reactivex.rxjava3.exceptions.UndeliverableException
Exception io.reactivex.rxjava3.exceptions.UndeliverableException:
  at io.reactivex.rxjava3.plugins.RxJavaPlugins.onError (RxJavaPlugins.java:372)
  at io.reactivex.rxjava3.internal.util.AtomicThrowable.tryAddThrowableOrReport (AtomicThrowable.java:52)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.innerError (FlowableFlatMap.java:564)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableFlatMap$InnerSubscriber.onError (FlowableFlatMap.java:636)
  at io.reactivex.rxjava3.internal.operators.single.SingleToFlowable$SingleToFlowableObserver.onError (SingleToFlowable.java:68)
  at io.reactivex.rxjava3.internal.operators.single.SingleDelayWithCompletable$OtherObserver.onError (SingleDelayWithCompletable.java:64)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableAndThenCompletable$SourceObserver.onError (CompletableAndThenCompletable.java:62)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableSubscribeOn$SubscribeOnObserver.onError (CompletableSubscribeOn.java:74)
[... cut ...]
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at java.lang.Thread.run (Thread.java:923)
Caused by com.polar.androidcommunications.api.ble.exceptions.BleAttributeError:
  at com.polar.androidcommunications.api.ble.model.gatt.BleGattBase.lambda$waitNotificationEnabled$1$com-polar-androidcommunications-api-ble-model-gatt-BleGattBase (BleGattBase.java:402)
  
[image]
[image]
[image]

Thanks!

JOikarinen, 2022-09-23:
Thanks a lot for the insight @biosignalssolutions, I will take a look for CRASH CASE 3 and CRASH CASE 4 as well. 

> It would be great to know **how/where to catch** these exceptions, so at least the app would continue to work.
- at least the CRASH CASE 1 was a bug within the SDK, which shall be handled by the SDK not by client application

biosignalssolutions, 2022-10-04:
Hi, 
Thank you for fix of CASE 1!!
I will incorporate it into my application and I will explain you how it work.

Have you figure out the issue of CASE 2?

JOikarinen, 2022-10-06:
Hi @biosignalssolutions, 

thanks for update. CASE2 to CASE4 looks like coming from same root cause. I have spent some time to figure out the exact root, but haven't pin point it yet. I will continue the investigations on this during coming days.

biosignalssolutions, 2022-10-18:
Hi Jukka,

I updated the sdk version to 3.3.3 and  minSdkVersion 24, but the exception still remain.
In older post you said you was able to reproduce the CASE 2. How can I reproduce it?

I noticed there is a new version of rxjava3, from 3.1.3 to 3.1.5
Should I update rxjava3 to 3.1.5 version ?

thanks,
Toni.

Exeception using  sdk version 3.3.3. Identical to CASE 2
Type
io.reactivex.rxjava3.exceptions.UndeliverableException
Exception io.reactivex.rxjava3.exceptions.UndeliverableException:
  at io.reactivex.rxjava3.plugins.RxJavaPlugins.onError (RxJavaPlugins.java:372)
  at io.reactivex.rxjava3.internal.util.AtomicThrowable.tryAddThrowableOrReport (AtomicThrowable.java:52)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.innerError (FlowableFlatMap.java:564)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableFlatMap$InnerSubscriber.onError (FlowableFlatMap.java:636)
  at io.reactivex.rxjava3.internal.operators.single.SingleToFlowable$SingleToFlowableObserver.onError (SingleToFlowable.java:68)
  at io.reactivex.rxjava3.internal.operators.single.SingleDelayWithCompletable$OtherObserver.onError (SingleDelayWithCompletable.java:64)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableAndThenCompletable$SourceObserver.onError (CompletableAndThenCompletable.java:62)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableSubscribeOn$SubscribeOnObserver.onError (CompletableSubscribeOn.java:74)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableCreate$Emitter.tryOnError (CompletableCreate.java:91)
  at com.polar.androidcommunications.api.ble.model.gatt.BleGattBase.lambda$waitNotificationEnabled$1$com-polar-androidcommunications-api-ble-model-gatt-BleGattBase (BleGattBase.java:417)
  at com.polar.androidcommunications.api.ble.model.gatt.BleGattBase$$ExternalSyntheticLambda1.subscribe
  at io.reactivex.rxjava3.internal.operators.completable.CompletableCreate.subscribeActual (CompletableCreate.java:40)
  at io.reactivex.rxjava3.core.Completable.subscribe (Completable.java:2859)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableSubscribeOn$SubscribeOnObserver.run (CompletableSubscribeOn.java:64)
  at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run (Scheduler.java:644)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run (ScheduledRunnable.java:65)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call (ScheduledRunnable.java:56)
  at java.util.concurrent.FutureTask.run (FutureTask.java:266)
  at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run (ScheduledThreadPoolExecutor.java:301)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at java.lang.Thread.run (Thread.java:923)
Caused by com.polar.androidcommunications.api.ble.exceptions.BleDisconnected:
  at com.polar.androidcommunications.api.ble.model.gatt.BleGattBase.lambda$waitNotificationEnabled$1$com-polar-androidcommunications-api-ble-model-gatt-BleGattBase (BleGattBase.java:404)

JOikarinen, 2022-10-23:
Hi Toni, sorry for delayed answer. The reproduce steps are like this, but the steps needs retries still to get error to happen. With example app the problem occurs like this:

Pre step:
- remove the Polar device pairing from the Android -> Bluetooth Settings 

Steps:
- Use "Auto Connect" or "Connect to <deviceId>" button to connect the device
- Android operating system pops up the pairing dialog, either ignore the pairing dialog or press cancel

Observed behavior:
- after the playing around for some time the PolarBleSDK library throws one of the CASE2-CASE4

I am quite confident the CASE2-CASE4 are all bugs in PolarBleSDK, I have some kind of clue what is happening in the SDK, but haven't been able to make the fix for the bugs yet. You may update the RxJava from 3.1.3 to 3.1.5 but I suspect the problem won't disappear.

biosignalssolutions, 2022-10-25:
Hi Jukka, I've tried causing the Execption, but no luck.

Reviewing the google play logs, I have observed another sensibly different exception, CASE 5. Maybe it can help you.
I'm going to publish another app version with RxJava 3.1.5. I'll inform you about it.

Thank you very much for your effort and dedication, I know that intermittent failures are very difficult to found.

Have a nice day.
Toni.

PD:  CASE 5
 Exception java.lang.NullPointerException:
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.GattCallback.onServicesDiscovered
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.BDDeviceListenerImpl$1.lambda$startServiceDiscovery$0$com-polar-androidcommunications-enpoints-ble-bluedroid-host-BDDeviceListenerImpl$1 (BDDeviceListenerImpl.java:349)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.BDDeviceListenerImpl$1$$ExternalSyntheticLambda0.run
  at io.reactivex.rxjava3.internal.observers.CallbackCompletableObserver.onComplete (CallbackCompletableObserver.java:44)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableObserveOn$ObserveOnCompletableObserver.run (CompletableObserveOn.java:89)
  at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run (Scheduler.java:644)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run (ScheduledRunnable.java:65)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call (ScheduledRunnable.java:56)
  at java.util.concurrent.FutureTask.run (FutureTask.java:264)
  at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run (ScheduledThreadPoolExecutor.java:307)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1137)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:637)
  at java.lang.Thread.run (Thread.java:1012)

JOikarinen, 2022-10-26:
Thanks Toni for sharing the information, very valuable. The problems you have reported (i.e. CASE1-CASE4) were new findings. The CASE5 I will take a look. 

I have kept couple investigation sessions with the cases2-4,  but unfortunately haven't yet been able to spent enough time on them to make the break through.

biosignalssolutions, 2023-02-03:
​No news.
At this moment we are working on how these exceptions could be caused (force), and how we can catch them.
​Do you know ​what is the purpose of the ​throwable block in the following code?

 api.requestStreamSettings(deviceId, PolarBleApi.DeviceStreamingFeature.ECG)
.toFlowable()
.flatMap((Function<PolarSensorSetting, Publisher<PolarEcgData>>) sensorSetting -> api.startEcgStreaming(deviceId, sensorSetting.maxSettings()))
.observeOn(AndroidSchedulers.mainThread())
.subscribe(
polarEcgData -> {
.........
},
​​**throwable -> {
//???
ecgDisposable = null;
},**
() -> Log.d(TAG_CLASS_NAME, " complete")
);

JOikarinen, 2023-02-09:
Guys, we haven't forget you and we are following this thread. At some point I thought I can reliably reproduce the issue, but I was too early on that statement. I roughly know the area where the problem is, but still the root cause remains mystery. 

Due to up coming 5.0.0 release there will be changes on API and also SDK is refacted behind the scenes. The refactoring will touch on the parts of the code where the problem is. 5.0.0 maybe won't fix the issue, but I hope we will get better insights what is going on if bug still appears.

biosignalssolutions, 2023-02-09:
Okay, Houston! we can verify some of readings for you.  ( Apollo 13 )

biosignalssolutions, 2023-03-09:
Hi,

Can you check the NullPointerException attached? Probably because an object is being accessed that its value is null. Perhaps this situation leaves the comunication in an unstable state.

On the other hand, there are new versions of rxjava3, maybe polar-ble-sdk library uses an old version. Or these libraries only are loaded from application build.gradle file?

implementation 'io.reactivex.rxjava3:rxjava:3.1.6'
implementation 'io.reactivex.rxjava3:rxandroid:3.0.2'

Thanks in advance.

**Exception java.lang.NullPointerException:**
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.GattCallback.onServicesDiscovered (GattCallback.kt:57)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.BDDeviceListenerImpl$1.lambda$startServiceDiscovery$0 (BDDeviceListenerImpl.java:345)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.BDDeviceListenerImpl$1.$r8$lambda$qOzZNKXSUCYPzl47WA3yMlElaQU (BDDeviceListenerImpl.java)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.BDDeviceListenerImpl$1$$InternalSyntheticLambda$2$ae3145d1f883817e8676d6511309840d3882e7f7e49da629a9015087f2a61b70$0.run (BDDeviceListenerImpl.java)
  at io.reactivex.rxjava3.internal.observers.CallbackCompletableObserver.onComplete (CallbackCompletableObserver.java:44)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableObserveOn$ObserveOnCompletableObserver.run (CompletableObserveOn.java:89)
  at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run (Scheduler.java:644)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run (ScheduledRunnable.java:65)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call (ScheduledRunnable.java:56)
  at java.util.concurrent.FutureTask.run (FutureTask.java:264)
  at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run (ScheduledThreadPoolExecutor.java:307)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1137)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:637)
  at java.lang.Thread.run (Thread.java:1012)

JOikarinen, 2023-03-16:
Hi @biosignalssolutions, on which version of the Polar BLE SDK you noticed the latest observation?

biosignalssolutions, 2023-03-16:
Hi,

In v3.3.3

Thank you,
Toni

JOikarinen, 2023-03-16:
Thanks Toni for the information. Can you reproduce the crash, or do you see it the crash in analytics? The area where the crash comes is enhanced in release v4.0.0. Are you able to migrate from Polar BLE SDK v3.3.3 to v4.0.0?

JOikarinen, 2023-03-28:
Finally some light on this thread. The cases 3 and 4 shall be fixed in release 5.0.1.

biosignalssolutions, 2023-03-30:
Hi!
Thank you very much for your perseverance on the issue.
It will take us a while to test the new version with various phones and users. Then we will give feedback.

We are glad to know that the project has continuous improvements ;-)

biosignalssolutions, 2023-06-29:
After several weeks testing no more crashes due OnErrorNotImplementedException, UndeliverableException
Thanks!

closed 2023-06-29 by biosignalssolutions