polarofficial/polar-ble-sdk #302: Keeping a Polar h10 active and transmitting while it is off the body

smyffanon, 2022-09-24:
***Platform your question concerns:*** 
- [x ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I would like to run a read while my h10 is out of the strap and off my body, but the unit turns off after a few seconds out of the strap. 

I'm hoping to get a more detailed exception report but when the read fails.  Even with only the phone, whatever is causing the fail takes a long time

Under logcat I think it's much, much slower  - could be more than a day & I can't be wearing the unit and carrying a phone and laptop.

JOikarinen, 2022-09-26:
Thanks for the report @smyffanon. 

Let's double check we understand the problem similar way. These are basically your steps:
- there is long recording saved in H10 memory (e.g. over night recording of heart rate or RR data)
- the recording read out fails, if the H10 is removed from the strap and strap is not fasten on user chest

JOikarinen, 2022-10-02:
@smyffanon sorry it took some time to get back on this issue. I can confirm the behaviour your described is an issue in H10. Even the exercise data read is started on H10 the H10 gives up the job after 45s if the H10 is removed from the strap and strap from the body. I will write the internal bug about this finding. For now there is no other workaround than just keeping the H10 in strap while reading the file.

closed 2022-10-02 by JOikarinen