polarofficial/polar-ble-sdk #303: Is it possible to query the battery level on demand?

orestesgaolin, 2022-09-25:
***Platform your question concerns:*** 
- [x] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

We noticed that the battery level may be reported with some quite high delay. Specifically when user puts their device to charge overnight, and neither it has been explicitly disconnected, nor the app has been closed. 
When the device reconnects with the app, it will not report the real battery level until it actually changes. This causes a problem on our side as the latest value the app is aware of is e.g. 10%. 
We were wondering if there's a way to query for battery level explicitly on reconnection?

Swift excerpt:

```swift
class PolarClient : PolarBleApiObserver,
                    PolarBleApiPowerStateObserver,
                    PolarBleApiDeviceInfoObserver,
                    PolarBleApiDeviceFeaturesObserver,
                    PolarBleApiDeviceHrObserver {
...

  init() {
    Logger.verbose(TAG, "Initialising.")
    
    polarBleApi = PolarBleApiDefaultImpl.polarImplementation(
      DispatchQueue.main,
      features: Features.allFeatures.rawValue
    );
    polarBleApi?.logger = PolarClientLogger(tag: TAG)
    
    polarBleApi?.observer = self
    polarBleApi?.deviceHrObserver = self
    polarBleApi?.powerStateObserver = self
    polarBleApi?.deviceFeaturesObserver = self
    polarBleApi?.deviceInfoObserver = self
  }

  var connectionListener: DeviceConnectionListener?
  func setConnectionListener(_ listener: DeviceConnectionListener) {
    Logger.verbose(TAG, "Setting connection listener")
    connectionListener = listener
  }
  
  var batteryListener: BatteryListener?
  func setBatteryListener(_ listener: BatteryListener) {
    Logger.verbose(TAG, "Setting battery listener")
    batteryListener = listener
  }

  func batteryLevelReceived(_ identifier: String, batteryLevel: UInt) {
    let battery = DeviceBattery(value: Int(batteryLevel), timestamp: Date())
    Logger.verbose(TAG, "Battery level updated: \(battery.toJsonString())")
    self.batteryListener?.onBattery(battery)
  }
```

If you take a look at the logs related to Polar connection you'll notice that the user tried to restart the app several times and each time the battery level reported was 10% despite the armband connected to charger overnight:

```
2022-09-25T07:50:40.116Z BFS: Initialising.
2022-09-25T07:50:40.224Z SBFSP: Restoring foreground service
[... 99 lines cut ...]
2022-09-25T07:53:38.319Z PC: Bluetooth turned off
2022-09-25T07:53:38.319Z PC: Setting heart rate listener
2022-09-25T07:53:38.319Z PC: Setting heart rate variability listener
2022-09-25T07:53:38.319Z PC: Setting battery listener
2022-09-25T07:53:38.319Z PC: Setting connection listener
2022-09-25T07:53:38.320Z PC: Connecting to device [REDACTED]
2022-09-25T07:53:38.332Z PC: Bluetooth turned on
2022-09-25T07:53:38.678Z PC: Device connection updated: {"connection": 0, "deviceId": "[REDACTED]", "name": "Polar Sense [REDACTED]" }
2022-09-25T07:53:51.134Z PC: Device connection updated: {"connection": 0, "deviceId": "[REDACTED]", "name": "Polar Sense [REDACTED]" }
2022-09-25T07:53:52.258Z PC: Device connection updated: {"connection": 1, "deviceId": "[REDACTED]", "name": "Polar Sense [REDACTED]" }
2022-09-25T07:53:52.260Z CSNR: Showing a notification that device got connected
2022-09-25T07:53:53.035Z PC: HR ready
2022-09-25T07:53:53.456Z PC: DIS info updated: Manufacturer Name String
2022-09-25T07:53:53.516Z PC: DIS info updated: Model Number String
2022-09-25T07:53:53.605Z PC: DIS info updated: Serial Number String
2022-09-25T07:53:53.665Z PC: DIS info updated: Hardware Revision String
2022-09-25T07:53:53.727Z PC: DIS info updated: Firmware Revision String
2022-09-25T07:53:53.876Z PC: DIS info updated: Software Revision String
2022-09-25T07:53:53.966Z PC: DIS info updated: System ID
2022-09-25T07:53:54.236Z PC: Battery level updated: {"value": 100, "timestamp": "1664092434236" }
```

At `07:52:49` the battery reported was 10%, whereas at `07:53:54` it was already 100%

Please let me know if you'd need any more insight into the implementation details or if you have any suggestion how to handle the battery level in more reliable way.

JOikarinen, 2022-09-28:
Hi @orestesgaolin, thanks for the very descriptive question. 

The battery level is always read from the device at the moment BLE connection is established. SDK makes the read similarly on Android and iOS, so both platforms shall behave the same. I did double check the SDK functionality. So I started to suspect this could be a device bug (i.e. bug in VeritySense firmware). What I found out, the VeritySense is  reporting the battery status it has when plugged to USB power source. So even the battery level is rising due the charging the VeritySense is still reporting the same charge status it had at the time it was plugged. 

For example:

1. VeritySense has low battery, e.g. 10%
2.  Plug in VeritySense to the charger
3. Wait for few minutes that charging has increased from  the 10%
4. Create BLE connection and see the battery level reported
    - Observation: Battery level 10%
5. Remove VeritySense from the charger and plug it back again
6. Create BLE connection and see the battery level reported -> 
    - now it correct, e.g. 50%

I have reported the issue internally in Polar. The issue will be triaged. I am about to close this issue  as the bug is not directly related to Polar BLE SDK.

closed 2022-09-28 by JOikarinen

JOikarinen, 2022-09-28:
Updated the [known issues](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/KnownIssues.md).

orestesgaolin, 2022-09-28:
Thank you! Is there any way to detect that the Polar Verity Sense is charging? That would allow us to ignore these readings.

JOikarinen, 2022-09-29:
@orestesgaolin unfortunately I cannot figure out any workaround to detect the charging, other than battery status is correctly reported by VeritySense.

orestesgaolin, 2022-09-30:
Interestingly I discovered today that the SDK throws exception when you try to subscribe to Accelerometer stream when the device is connected to charger

```
  (23771): ACC stream failed
E/PC      (23771): com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command error:  failed with error: ERROR_DEVICE_IN_CHARGER
E/PC      (23771): 	at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient.sendControlPointCommand$lambda-14(BlePMDClient.kt:246)
E/PC      (23771): 	at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient.$r8$lambda$_mWIojlvs7ItI9lsdIAR6dRGbxQ(Unknown Source:0)
E/PC      (23771): 	at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient$$ExternalSyntheticLambda1.subscribe(Unknown Source:6)
E/PC      (23771): 	at io.reactivex.rxjava3.internal.operators.single.SingleCreate.subscribeActual(SingleCreate.java:40)
E/PC      (23771): 	at io.reactivex.rxjava3.core.Single.subscribe(Single.java:4855)
E/PC      (23771): 	at io.reactivex.rxjava3.internal.operators.single.SingleSubscribeOn$SubscribeOnObserver.run(SingleSubscribeOn.java:89)
E/PC      (23771): 	at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run(Scheduler.java:644)
E/PC      (23771): 	at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:65)
E/PC      (23771): 	at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:56)
E/PC      (23771): 	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
E/PC      (23771): 	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:307)
E/PC      (23771): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
E/PC      (23771): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
E/PC      (23771): 	at java.lang.Thread.run(Thread.java:1012)
```

Wondering if this could be detected in a bit more graceful way?

JOikarinen, 2022-10-01:
@orestesgaolin I see you concern here. Have you implemented the onError handler for [startAccStreaming](https://github.com/polarofficial/polar-ble-sdk/blob/4ae25e30f7223a14a1613fe17e7328f1ae1936dc/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L413) subscriber? Something like:
```
api.startAccStreaming(deviceId, settings)
    .subscribe(
      { polarAccelerometerData: PolarAccelerometerData ->
      },
      { error: Throwable -> 
          Log.e(TAG, "ACC stream failed. Reason $error")
      },
      {
          Log.d(TAG, "ACC stream complete")
      }
)
 ```
 
 At the moment I cannot figure out better way to share the details why the ACC couldn't be started. Do you have better ideas how the cause of not able to start the ACC shall be told for the user of the API?

orestesgaolin, 2022-10-01:
Hey there, just to clarify - I have enabled the error handling and I logged the exception intentionally. What I was trying to ask about is whether it's possible to get information about Verity Sense in charger assuming:

1. The band clearly knows when it's charging as it does not allow to stream ACC data
2. Seems like the SDK is able to recognize this error and throw it as `ERROR_DEVICE_IN_CHARGER`
3. It feels like it should be possible to get the same information not only when the ACC stream is requested, but also at any other time, but maybe I'm wrong and it's far from achievable

JOikarinen, 2022-10-02:
Good thoughts. 👍 The charger status shall be pretty trivial to read from device, but that simple task which is  also simple to implement needs surprisingly some work effort. 

Battery status is read from the device using the standard [BLE Battery service](https://www.bluetooth.com/specifications/specs/battery-service-1-0/), unfortunately the Battery service specification don't have support for charger status (e.g. isCharging, isPluggedToCharger or anything similar). Thus the custom BLE service to indicate the charger status is needed to be implement on device side, along with the corresponding client on mobile side. For now I cannot promise much related to this feature, but definitely I will create the internal feature request on this.

The reason why the ```startAccStream()``` can return the ```ERROR_DEVICE_IN_CHARGER``` status is because the answer is returned by the device when Acc start is requested.

ranjitIN, 2023-06-12:
Hey there, just to clarify is there any way to read the device in charge or not without connecting to it. like in advertisement data or something else in which i can get the device status is my device in charge or not.

jimmyzumthurm, 2023-07-06:
Hi @ranjitIN 
Unfortunately there is no way of getting the battery status of device without connecting to it.

ranjitIN, 2023-07-06:
Thanks for updating.