polarofficial/polar-ble-sdk #304: Is it possible to detect no skin contact using Polar Verity Sense

orestesgaolin, 2022-09-27:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

We got several reports from our users that the Polar Verity Sense is streaming the HR data despite being off their arm or when flipped backwards (so that the sensor points away from the skin).

We tried to test it by ourselves and it seems that when the Verity Sense is flipped, it will still report HR, and this seems to be affected by the most recent value. Interestingly, the value is also reported if you don't put the Verity Sense on your arm at all. I noticed that when the band is not in contact with the skin the data is reported much more sparsely. 

To sum up:

- band on arm, correct orientation - reporting data
- band on arm, incorrect orientation, sensor flipped - reporting data; I noticed that Polar Flow app is a little bit better in ignoring the data reported by the SDK
- band off arm - reporting data with breaks

We would love to learn if it's possible to ignore the HR values when the sensor is not in contact with the skin. Maybe there's some characteristics of the "idle" data that we could take into account to manually ignore the readings? Do you have any recommendations how to handle uncertainty of the data? Does getting `0` as heart rate value can be considered as indication of lack of contact?

JOikarinen, 2022-09-28:
Hi @orestesgaolin,

I understand your concerns and questions. The contactStatusSupported bit in [PolarHrData](https://github.com/polarofficial/polar-ble-sdk/blob/7810974560d9d6c23d7b9f9863eb2563eaf73e01/sources/Android/android-communications/src/main/java/polar/com/sdk/api/model/PolarHrData.java#L36) is false for Polar OH1 and for Polar Verity Sense. That is because the detection of the skin contact is challenging to implement reliably. If the optical heart sensor is not contacting the skin it is likely that optical measurement unit gets light from the surrounding environment. That is why you see random values once the VeritySense is removed from the arm or turned around.

As that being said I don't have good advices for you problem. Maybe a logic written to your app which detect sudden changes in heart rate values could be an option.

closed 2022-09-28 by JOikarinen

JOikarinen, 2022-09-28:
> 4\. Is there anything else taken into account when calculating the HR besides optic sensor? Maybe acc or gyro?
- quote from the [Polar white paper](https://www.polar.com/sites/default/files/static/science/white-papers/polar-optical-heart-rate-white-paper.pdf): 
"The optical measurement is very sensitive to motion[5]. 1 millimeter movement in the distance from the
sensor to the skin will totally deteriorate the pulse wave signal. Without adequate and efficient motion artefact removal the pulse wave would be unrecognizable and heart rate would not be available.

The compensation of motion artefacts is the most important and most demanding part of the signal processing of Wrist Heart Rate measurement. The compensation is based on a motion sensor located in the wrist unit and measuring very accurately the movements of the hand. The signal processing algorithms for Polar Wrist Heart Rate measurement are proprietary to Polar."

orestesgaolin, 2022-09-28:
Thanks a lot for the answers. 

I was wondering if for the case when the band is clearly off the arm, would there be a way to deduce that from accelerometer and gyro readings showing 0 in all axes? Could we use it as some indicator of no movement or the noise would be indistinguishable from the person just sitting steadily?

Yaro-o, 2023-04-26:
> 

Hey @orestesgaolin, I am looking into the same problem. Did you get anywhere with this task?

orestesgaolin, 2023-04-26:
Unfortunately no, we were focused on the stability improvements and it seems that users don't notice this particular issue so much as we do