polarofficial/polar-ble-sdk #305: Polar Bluetooth Wearlink packet

duoduodady, 2022-09-28:
I used "nRF Connect"  to Connect Polar H7.

[image]

The Raw data:
0x02010603020D1807FF6B003B1B5452020A001209506F6C6172204837204437393034433134

The third part Hex is Very important!
"07FF6B003B1B5452"

Format  result:
07 FF 6B 00 3B 1B 54 52

I guess the format Info is :
- 6B 00  :  The Company
- 3B   ????    I don't konw
- 1B   ????    I don't konw
- 54   ????   HeartRate  ???
- 52   ?????  RRInterval ???

Somebody can help me ?

zHElEARN, 2023-01-23:
这里是不包含HR和RR interval信息的，你要读取这些信息要从00002a37-0000-1000-8000-00805f9b34fb读取

JOikarinen, 2023-01-26:
I am closing this issue as not related to SDK.

closed 2023-01-26 by JOikarinen