polarofficial/polar-ble-sdk #307: SDK functionalities to configure the polar H10 with several features

biosignalssolutions, 2022-10-05:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [X] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [X] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

Hello,

It would be great to add SDK functionalities to configure the polar H10 with the follow features:

- Select sample rate until 260Hz, despite the fact that this supposes an increase in battery consumption.
- Select the gain of the digital converter, bettween x0.5, x1, x2
- Select the cutoff frequencies of the high pass digital filter (maybe this is not possible because the filters are by hardware).

Are these features possible, or its too hard to implement?

Thanks you,

JOikarinen, 2022-10-06:
Thanks for the proposals @biosignalssolutions. I will proceed with the suggestions and bring them forward. 

**260Hz sample rate for ECG :** I guess you mean the ECG, not ACC.  Could you explain a bit more detailed, why this would particularly important for your use cases? It would help me better argument the need.

**Gain of the digital converter, between x0.5, x1, x2 and Select the cutoff frequencies of the high pass digital filter**: Could you explain a bit more detailed, why this would particularly important for your use cases? It would help me better argument the need.

biosignalssolutions, 2022-10-06:
Yes, I mean only for the ECG for Polar H10 device.

-**260Hz sample rate for ECG (in fact, it is twice the actual sample rate)**: 
Argument 1)
To display the signal in programs such as EDFBrowser, a minimum frequency of 250Hz is required.

Argument 2)
We're trying to measure the QT interval in milliseconds. Right now, at the current sample rate, we have a resolution of about 8 ms. For our graphs it represents a variation of 16 ms, and we would like to minimize our error. (See QT chart). By multiplying the sample rate by 2, we increase the resolution by 2 (4ms) and hopefully halve the error.
[image]

**-Select the cutoff frequencies of the high pass digital filter**:
As a result of the filters there is a small deformation at the end of the T wave. I suppose that you have the cutoff frequency around 0.5Hz to remove baseline wander. We expect to reduce T wave small deformation by reducing cutoff frequency.
[image]

**- Gain of the digital converter, between x0.5, x1, x2** 
As a result of reduce the cutoff freqüency it will apear baseline wander, and to avoid the saturation of analog converter, one way is reduce the gain.

**For us the most important thing is to increase the sample rate**. Proposals of cutoff frequency and gain are less important.
I recognize that we are asking for some characteristics that maybe go beyond the objective of the product. But at the same time, the ability to read the ECG of the H10 and SDK, **makes your product unique and powerful!**.

Thanks.

JOikarinen, 2022-10-08:
@biosignalssolutions  Thanks a lot for very good arguments. Each of the feature requests are separately forwarded. As each of request is related to H10 product, not exactly to SDK, so I will close this issue for now. If there will be insights I can share, I will give an update.

closed 2022-10-08 by JOikarinen