polarofficial/polar-ble-sdk #309: No longer receiving `streamingFeaturesReady` on iOS

Rexios80, 2022-10-11:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
- `streamingFeaturesReady` is not called
- Still works on Android
- Broken on version 3.3.4
- Works on version 3.3.3

**How to Reproduce**
Connect a device and listen for a `streamingFeaturesReady` call

**Expected behavior**
It gets called

**Screenshots and logs**
I don't see anything useful in the logs

JOikarinen, 2022-10-11:
Thanks for observation @Rexios80. I suspect your finding has something to do with [Bluetooth State Preservation and Restoration](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW10:~:text=and%20pending%20connections.-,State%20Preservation%20and%20Restoration,-Because%20state%20preservation) introduced in version 3.3.4.

I tried to reproduce the issue with different kinds of steps with [example app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-ios/iosBleSdkTestApp), but I couldn't yet reproduce. Do you have some ideas how shall I try to reproduce the issue? Here is what I did:

STEPS:
1. Connect to H10 (used both ways of connect methods Auto Connect and Connect buttons)
2. Followed the `streamingFeaturesReady` callback

OBSERVATION:
- `streamingFeaturesReady` callback was called with ECG and ACC being ready

STEPS:
1. Connect to H10
2. Moved the H10 out of range
3. Bring H10 back to range -> example app automatically connects to H10

OBSERVATION:
- `streamingFeaturesReady` callback was called with ECG and ACC being ready

STEPS:
1. Connect to H10
2. Long press the power button and then long press home button to simulate the situation operation system terminates the app

OBSERVATION:
- connection with H10 was restored 
- `streamingFeaturesReady` callback was called with ECG and ACC being ready

Rexios80, 2022-10-11:
In my project I have tested where the only changes are the version of the cocoapod. I have tried upgrading the code that works (from before I upgraded the cocoapod) and downgrading the current code that doesn't work and I get the same results.

Rexios80, 2022-10-11:
Here is the repo for my project, except I'm not sure how helpful it will be since it's a Flutter plugin: https://github.com/Rexios80/polar

JOikarinen, 2022-10-11:
@Rexios80 Thanks for the additional information. Btw, you are working with interesting project!! 

I played around a bit with your project, thanks to you project I found the issue. I am able to reproduce the problem with the example app too:

STEPS:

- Connect to H10 
- see `streamingFeaturesReady` is called with with ECG and ACC being ready
- Start either ACC or ECG stream
- Disconnect the H10
- Connect to H10

OBSERVATION:
- after the last step the  `streamingFeaturesReady` is never called again even connecting and disconnecting.

ANALYSIS:
- the problem is indeed related to [Bluetooth State Preservation and Restoration](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW10:~:text=and%20pending%20connections.-,State%20Preservation%20and%20Restoration,-Because%20state%20preservation) introduced in version 3.3.4.

I will start investigations for the fix. Sorry for the trouble.

JOikarinen, 2022-10-11:
Fixed in Release 3.3.5

JOikarinen, 2022-10-11:
Reopen because there might be still problems in Release 3.3.5. I will come back on this as soon as possible.

JOikarinen, 2022-10-12:
The release 3.3.4 introduced the regression reported in this issue. The release 3.3.5 did not fully fix regression. But I hope the release 3.3.6 will do. 

Apologies for the extra hassle on this.

closed 2022-10-12 by JOikarinen