polarofficial/polar-ble-sdk #311: Too many Processing service data. Status: 0.  Data length: 2 logs

kiolep, 2022-10-18:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I'm writing an app, where a Verity Sense and a H10 device connects to phone. When both of them are connected, they start to stream data. From the H10, I want to collect HR and RRms data, and from the VS, I want to collect HR and PPI data. 
I get what I need from the H10 from the `hrNotificationReceived`, and for the VS I just use the code below in the OnCreate method.

```
            val isDisposed = ppiDisposable?.isDisposed ?: true
            if (isDisposed || deviceConnected) {
                ppiDisposable = api.startOhrPPIStreaming(VSid)
                    .observeOn(Schedulers.computation())
                    .subscribe(
                        { ppiData: PolarOhrPPIData ->
                            for (sample in ppiData.samples) {
                                Log.d(TAG, "VS: HR: ${sample.hr} // PPI: ${sample.ppi}")
                                PPI_write.text = sample.ppi.toString()
                                PPG_write.text = sample.hr.toString()
                            }
                        },
                        { error: Throwable ->
                            Log.e(TAG, "PPI stream failed. Reason $error")
                        },
                        { Log.d(TAG, "PPI stream complete") }
                    )
            } else {
                ppiDisposable?.dispose()
            }
```

I have no problem with the H10, but the VS doesn't log the data I would expect it to log. I have 2 main problems with it.
	
1. I noticed, that a lot of ```BleHrClient/Processing service data. Status: 0.  Data length: 2``` log turns up. I suspect that these are data received from the VS sensor, that are for some reason are not loged out properly by the Log.d(). Also with these, I noticed, that I receive a lot of these in a 5 sec period and then suddenly for like 15ms properly logged data arrives. After that again 5 second of "Processing service data", and again 15ms of good data.
	
	What does this ```BleHrClient/Processing service data. Status: 0.  Data length: 2``` mean?
	
	Why might this 5sec of no data, then 15ms of good data rotation occures? What am I missing? Is there some bad practice I'm using?

	What is this  ```No factor found for type: PPI ``` I found a similar issue at #186, but that didn't help.
	

	
2. My second question is, sometime when I get proper data from Verity Sense, the HR value fuctuates a great extent. Sometimes less noticable (it goes up from 55BPM to 60BP within 5sec while I do nothing, and then down to 53BPM 5sec later), sometimes it's basically unusable data (from 60BPM to 0BPM and then again back to 60BPM).
	
	Is this maybe due to improper contact with the sensor, or the sensor has some buffer, that slowly empties itself?
	
Logs included below (From where I think could be interesting. Also note, I only use VS in this log):
```
I/DecorView: [INFO] isPopOver=false, config=true
I/DecorView: updateCaptionType >> DecorView@316418b[], isFloating=false, isApplication=true, hasWindowDecorCaption=false, hasWindowControllerCallback=true
D/DecorView: setCaptionType = 0, this = DecorView@316418b[]
D/Streams: BluetoothStateChanged true
D/InputTransport: Input channel constructed: '476c15f', fd=89
I/ViewRootImpl@68806f3[StreamsActivity]: setView = com.android.internal.policy.DecorView@316418b TM=true
I/SurfaceControl: assignNativeObject: nativeObject = 0 Surface(name=null)/@0xc206f0c / android.view.SurfaceControl.readFromParcel:1117 android.view.IWindowSession$Stub$Proxy.relayout:1820 android.view.ViewRootImpl.relayoutWindow:9005 android.view.ViewRootImpl.performTraversals:3360 android.view.ViewRootImpl.doTraversal:2618 android.view.ViewRootImpl$TraversalRunnable.run:9971 android.view.Choreographer$CallbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 
I/ViewRootImpl@68806f3[StreamsActivity]: Relayout returned: old=(0,0,1080,2400) new=(0,0,1080,2400) req=(1080,2400)0 dur=14 res=0x7 s={true -5476376604669168336} ch=true fn=-1
I/ViewRootImpl@1d424e9[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
I/SurfaceControl: nativeRelease nativeObject s[-5476376608964095792]
I/SurfaceControl: nativeRelease nativeObject e[-5476376608964095792]
[... 168 lines cut ...]
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/API LOGGER: BlePMDClient/pmd data:  HEX: 03 00 00 00 00 00 00 00 00 00 3C 2C 04 06 00 06 3C 09 04 06 00 06 3C A2 03 07 00 06 3C 82 03 09 00 06 3C 23 04 0A 00 06 
D/API LOGGER: BlePMDClient/No factor found for type: PPI
D/Streams: VS: HR: 60 // PPI: 1068
D/Streams: VS: HR: 60 // PPI: 1033
D/Streams: VS: HR: 60 // PPI: 930
D/Streams: VS: HR: 60 // PPI: 898
D/Streams: VS: HR: 60 // PPI: 1059
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/API LOGGER: BlePMDClient/pmd data:  HEX: 03 00 00 00 00 00 00 00 00 00 39 8D 04 09 00 06 39 5B 04 07 00 06 39 FB 03 07 00 06 39 29 04 07 00 06 39 34 04 06 00 06 
D/API LOGGER: BlePMDClient/No factor found for type: PPI
D/Streams: VS: HR: 57 // PPI: 1165
D/Streams: VS: HR: 57 // PPI: 1115
D/Streams: VS: HR: 57 // PPI: 1019
D/Streams: VS: HR: 57 // PPI: 1065
D/Streams: VS: HR: 57 // PPI: 1076
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
```

kiolep, 2022-10-21:
My main issue is probably exactley like #66

JOikarinen, 2022-10-24:
Hi @kiolep, thanks for the questions, I guess I understood the problems you are facing. Let's discuss them.

> I get what I need from the H10 from the `hrNotificationReceived`
- perfect

> 1. I noticed, that a lot of `BleHrClient/Processing service data. Status: 0.  Data length: 2` log turns up
- These are heart rate notifications received by the standard BLE Hr Client implemented in the SDK. In case of the H10 you are listening them by using `hrNotificationReceived` callback. However, in case of the VeritySense I expect that you are listening the PPI stream only, i.e. `api.startOhrPPIStreaming(VSid)`. 
- With VeritySense you may also listen the `hrNotificationReceived` to get heart rate update once a second

> 1. What is this  `No factor found for type: PPI ` I found a similar issue at [ Failed with error: ERROR_ALREADY_IN_STATE #186](https://github.com/polarofficial/polar-ble-sdk/issues/186), but that didn't help.
- how often you get this and do you have specific steps of this scenario? There is something small I see in the code snippet you shared, could be slightly related:
```
...
 val isDisposed = ppiDisposable?.isDisposed ?: true
             // isDisposed OR device is connected -> the problem is that if "isDisposed == false" the new stream can be started even the old stream still up and running. 
            if (isDisposed || deviceConnected) {
...            
```
shall be
 ```
...
 val isDisposed = ppiDisposable?.isDisposed ?: true
            // isDisposed AND device is connected
            if (isDisposed && deviceConnected) {
...            
```

> 2\. My second question is, sometime when I get proper data from Verity Sense, the HR value fuctuates a great extent. Sometimes less noticable (it goes up from 55BPM to 60BP within 5sec while I do [...]

- this is the problem in VeritySense itself and reported internally as problematic for SDK users.  The reason for 0 HR is that when enabling the PPI stream the firmware switches on different heart rate calculation algorithm. The reason why algorithm is changed is that Pulse to Pulse interval (PPI) calculation is much more sensitive for any movements of the device or other possible interference than the heart rate algorithm. When the PPI is streamed the heart rate value becomes unstable in `hrNotificationReceived` callback too. Here is the collection of explanations from previous issues:

https://github.com/polarofficial/polar-ble-sdk/issues/197#issuecomment-927098699
https://github.com/polarofficial/polar-ble-sdk/issues/197#issuecomment-949287862

kiolep, 2022-10-24:
Hey @JOikarinen, thanks for helping me!

I fixed the mistake in my code that you pointed out, but unfortunateley didn't solve my problem. My problem still remains that I don't get continous PPI values from VS. I get a batch of PPI data for 15ms, and then no data for 5s (see log snippet below). This problem also appears for me when I use the androidBleSdkTestApp.

As you can also see i get the 'No factor found for type: PPI' log in the beginning of each of these "batches". (Don't mind the VS_HR values, those are from the `hrNotificationReceived`)

```
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BlePMDClient/pmd data:  HEX: 03 00 00 00 00 00 00 00 00 00 00 19 01 1E 00 07 00 75 03 1E 00 07 00 B6 01 1E 00 07 00 5E 02 1E 00 06 00 94 02 1E 00 06 
D/API LOGGER: BlePMDClient/No factor found for type: PPI
D/API LOGGER: BlePMDClient/pmd data:  HEX: 03 00 00 00 00 00 00 00 00 00 00 17 02 1E 00 06 00 57 02 1E 00 06 00 57 02 1E 00 06 00 69 02 1E 00 06 00 7F 02 1E 00 06 
D/API LOGGER: BlePMDClient/No factor found for type: PPI
D/Streams: TIME: 2022-10-24T08:14:18.622Z VS: HR: 0 // PPI: 281
D/API LOGGER: BlePMDClient/pmd data:  HEX: 03 00 00 00 00 00 00 00 00 00 00 82 02 1E 00 06 00 6C 02 1E 00 06 00 72 02 1E 00 06 00 77 02 1E 00 06 
D/API LOGGER: BlePMDClient/No factor found for type: PPI
D/Streams: TIME: 2022-10-24T08:14:18.654Z VS: HR: 0 // PPI: 885
D/Streams: TIME: 2022-10-24T08:14:18.657Z VS: HR: 0 // PPI: 438
D/Streams: TIME: 2022-10-24T08:14:18.660Z VS: HR: 0 // PPI: 606
D/Streams: TIME: 2022-10-24T08:14:18.662Z VS: HR: 0 // PPI: 660
D/Streams: TIME: 2022-10-24T08:14:18.664Z VS: HR: 0 // PPI: 535
D/Streams: TIME: 2022-10-24T08:14:18.665Z VS: HR: 0 // PPI: 599
D/Streams: TIME: 2022-10-24T08:14:18.666Z VS: HR: 0 // PPI: 599
D/Streams: TIME: 2022-10-24T08:14:18.667Z VS: HR: 0 // PPI: 617
D/Streams: TIME: 2022-10-24T08:14:18.669Z VS: HR: 0 // PPI: 639
D/Streams: TIME: 2022-10-24T08:14:18.670Z VS: HR: 0 // PPI: 642
D/Streams: TIME: 2022-10-24T08:14:18.671Z VS: HR: 0 // PPI: 620
D/Streams: TIME: 2022-10-24T08:14:18.673Z VS: HR: 0 // PPI: 626
D/Streams: TIME: 2022-10-24T08:14:18.674Z VS: HR: 0 // PPI: 631
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/Streams: Battery level A4C84121 90%
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BlePMDClient/pmd data:  HEX: 03 00 00 00 00 00 00 00 00 00 00 88 02 1B 00 06 00 94 02 16 00 06 00 86 02 13 00 06 00 9B 02 11 00 06 00 A0 02 0E 00 06 
D/API LOGGER: BlePMDClient/No factor found for type: PPI
D/Streams: TIME: 2022-10-24T08:14:23.652Z VS: HR: 0 // PPI: 648
D/Streams: TIME: 2022-10-24T08:14:23.657Z VS: HR: 0 // PPI: 660
D/Streams: TIME: 2022-10-24T08:14:23.660Z VS: HR: 0 // PPI: 646
D/Streams: TIME: 2022-10-24T08:14:23.664Z VS: HR: 0 // PPI: 667
D/API LOGGER: BlePMDClient/pmd data:  HEX: 03 00 00 00 00 00 00 00 00 00 00 A6 02 0D 00 06 00 A9 02 0C 00 07 00 AF 01 0F 00 07 00 FB 01 13 00 07 
D/API LOGGER: BlePMDClient/No factor found for type: PPI
D/Streams: TIME: 2022-10-24T08:14:23.668Z VS: HR: 0 // PPI: 672
D/Streams: TIME: 2022-10-24T08:14:23.672Z VS: HR: 0 // PPI: 678
D/Streams: TIME: 2022-10-24T08:14:23.675Z VS: HR: 0 // PPI: 681
D/Streams: TIME: 2022-10-24T08:14:23.678Z VS: HR: 0 // PPI: 431
D/Streams: TIME: 2022-10-24T08:14:23.681Z VS: HR: 0 // PPI: 507
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BlePMDClient/pmd data:  HEX: 03 00 00 00 00 00 00 00 00 00 00 C3 01 11 00 07 00 86 01 11 00 07 00 7A 01 12 00 07 00 69 01 11 00 07 00 4E 02 11 00 07 
D/API LOGGER: BlePMDClient/No factor found for type: PPI
D/Streams: TIME: 2022-10-24T08:14:28.447Z VS: HR: 0 // PPI: 451
D/Streams: TIME: 2022-10-24T08:14:28.453Z VS: HR: 0 // PPI: 390
D/Streams: TIME: 2022-10-24T08:14:28.457Z VS: HR: 0 // PPI: 378
D/API LOGGER: BlePMDClient/pmd data:  HEX: 03 00 00 00 00 00 00 00 00 00 00 58 02 11 00 06 00 71 02 0F 00 06 00 8D 02 0C 00 06 00 91 02 0B 00 06 
D/API LOGGER: BlePMDClient/No factor found for type: PPI
D/Streams: TIME: 2022-10-24T08:14:28.460Z VS: HR: 0 // PPI: 361
D/Streams: TIME: 2022-10-24T08:14:28.465Z VS: HR: 0 // PPI: 590
D/Streams: TIME: 2022-10-24T08:14:28.469Z VS: HR: 0 // PPI: 600
D/Streams: TIME: 2022-10-24T08:14:28.473Z VS: HR: 0 // PPI: 625
D/Streams: TIME: 2022-10-24T08:14:28.476Z VS: HR: 0 // PPI: 653
D/Streams: TIME: 2022-10-24T08:14:28.479Z VS: HR: 0 // PPI: 657
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
D/Streams: VS_HR: 0
D/API LOGGER: BleHrClient/Processing service data. Status: 0.  Data length: 2
```

JOikarinen, 2022-10-25:
> My problem still remains that I don't get continous PPI values from VS. I get a batch of PPI data for 15ms, and then no data for 5s (see log snippet below). This problem also appears for me when I use the androidBleSdkTestApp.

The PPI data arrives in patches from VeritySense, so in one received packet you see multiple measured PPI values. Here is part of your log, lets analyse it
```
D/Streams: TIME: 2022-10-24T08:14:23.652Z VS: HR: 0 // PPI: 648
D/Streams: TIME: 2022-10-24T08:14:23.657Z VS: HR: 0 // PPI: 660
D/Streams: TIME: 2022-10-24T08:14:23.660Z VS: HR: 0 // PPI: 646
D/Streams: TIME: 2022-10-24T08:14:23.664Z VS: HR: 0 // PPI: 667
```

The Peak-to-Peak interval (PPI) is milliseconds between detected Peak pulses. So in above logs the received data covers time span of 648ms+660ms+646ms+667ms = 2621ms 

> As you can also see i get the 'No factor found for type: PPI' log in the beginning of each of these "batches". (Don't mind the VS_HR values, those are from the `hrNotificationReceived`)
- the "No factor found for type: PPI" is Polar BLE SDK internal log and that's nothing to be worried about . In my previous reply I got confused because there has been a bug where the exception is thrown saying "no factor found". Sorry about the confusion.

kiolep, 2022-10-25:
Ohh of course! I got confused by the time of the _logging_ of the data, and didn't think through further. Then everything works as intended.

Thanks a lot for helping me @JOikarinen! You saved me from a lot of further head scratches.

closed 2022-10-25 by kiolep