polarofficial/polar-ble-sdk #312: very large numbers in rr recording

smyffanon, 2022-10-20:
***Platform your question concerns:*** 
- [x ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Am I mis-using my h10 or is this something that happens once in a while?
Or is the unit beginning to break down? 
On two recent recordings I have gotten his type of output near the beginning of the file - around 200 lines down out of 30,000 lines.

893
1878949389
0
28
14679420
0
28
23068028
0
26
18873724
0
27
68732
805
888
898
902
884

JOikarinen, 2022-10-24:
Hi @smyffanon, that looks suspicious.  I don't believe the unit itself is breaking down, but strap might get old if you have used it a lot. You mentioned the problem happens beginning of recording, let's double check two things before diving deeper into the problem:

Firstly is the H10 moisten and strap steadily placed: https://support.polar.com/e_manuals/h10-heart-rate-sensor/polar-h10-user-manual-english/wear-the-heart-rate-sensor.htm

Secondly have you changed the H10 battery lately: https://support.polar.com/e_manuals/h10-heart-rate-sensor/polar-h10-user-manual-english/batteries.htm 

Btw, are you using online  recording (i.e. real time streaming over BLE) or offline recording into H10 memory?

smyffanon, 2022-10-25:
I did not suspect the strap, I have several so I'll have to track the next time this happens. 

moistening - yes, I do moisten always.  

Polar beat is showing the battery as full - I replaced it around 6 weeks ago. 

the recording I am talking about is on the unit's memory, not BLE.

closed 2025-07-03 by rkangast