polarofficial/polar-ble-sdk #313: Unable to connect to H10 following another app connecting and then terminating on Android

theMagicalTree, 2022-10-29:
**Platform on which you observed the bug:** 
- [ ] Android 12

**Device on which you observed the bug:**
- [ ] Polar H10

**Describe the bug**
The call to PolarBleApi.connectToDevice(deviceId) does not result in successful connection following another app on the same phone (Elite HRV) connecting and then terminating (including system level forced closure of app). Prior to use of the other app connection works fine.

**How to Reproduce**
If you know how the bug may be reproduced please add the steps here

1. Connect successfully to device using PolarBleApi.connectToDevice(deviceId)
2. Disconnect and close down app
3. Start Elite HRV and connect to device
4. Disconnect from device and terminate app
5. Attempt to connect as in 1. but call to connect never results in connection callback

**Expected behavior**
Call to callback function deviceConnected(polarDeviceInfo: PolarDeviceInfo) is never called. The app is left waiting

**Screenshots and logs**

2022-10-29 14:54:10.683 16138-16138/com.nickDTaylor.cardioCapture I/ViewRootImpl@5d71f21[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
2022-10-29 14:54:10.696 16138-16138/com.nickDTaylor.cardioCapture D/InputMethodManager: startInputInner - Id : 0
2022-10-29 14:54:10.697 16138-16138/com.nickDTaylor.cardioCapture I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
2022-10-29 14:54:12.522 16138-16138/com.nickDTaylor.cardioCapture I/ViewRootImpl@5d71f21[MainActivity]: ViewPostIme pointer 0
2022-10-29 14:54:12.609 16138-16138/com.nickDTaylor.cardioCapture I/ViewRootImpl@5d71f21[MainActivity]: ViewPostIme pointer 1
2022-10-29 14:54:12.614 16138-16138/com.nickDTaylor.cardioCapture D/Connection: Attempting connection to device 9F6CCE20
2022-10-29 14:54:12.617 16138-16138/com.nickDTaylor.cardioCapture D/Connection: API before connecting is com.polar.sdk.impl.BDBleApiImpl@a1dbf90
2022-10-29 14:54:12.618 16138-16138/com.nickDTaylor.cardioCapture D/Connection: AFter requesting connection

Many thanks,

Nick

theMagicalTree, 2022-10-29:
If bluetooth is disable then re-enabled at the system level connecting to the device works fine.

JOikarinen, 2022-10-31:
Hi @theMagicalTree,

thanks for the feedback. I tried to repeat with similar steps, by using [Polar BLE SDK example app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp) and EliteHRV. Both installed in the same phone. 

My Steps:

1. Polar BLE Example app: Connect with H10 -> Connection succeeds
2. Polar BLE  Example app: Disconnect the H10
3. EliteHRV: User profile -> Devices -> H10 is found -> press "+" to connect -> Connection created
4. EliteHRV: Terminate the app
5. Polar BLE Example app: Connect with H10 -> Connection succeeds

Observation:
- connection works as intended

Am I missing something from the steps? Do you have possibility to try with other Android phone?

stuartlynne, 2022-11-08:
The "other app" is not correctly closing the BLE connection.

You might be able to mitigate this by enabling the second BLE connection on the H10 (use the Polar Beat app.)

theMagicalTree, 2022-11-09:
Thank you Stuart, this has in fact resolved the issue.

I had previously seen this setting but thought it irrelevant as there is only a single device connecting to the H10, but with multiple apps. Perhaps this setting needs to be renamed?

JOikarinen, 2022-11-14:
Thanks @stuartlynne for help on this. In addition to Polar Beat, the second BLE connection can be enabled by the FlowApp too. Here are the instructions for both apps (Beat and FlowApp) https://support.polar.com/en/support/can_i_pair_h10_with_several_devices

I will close the issue as resolved, please don't hesitate to reopen.

closed 2022-11-14 by JOikarinen