polarofficial/polar-ble-sdk #317: Verity Sense inconsistent ACC Streaming

ymatt345, 2022-11-21:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [x] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

I am working on the additional  support of Verity Sense to my App that already supports OH1.
ACC stream datas from OH1 come every second with 36 samples.
However ACC stream datas from VeritySense comes every second or 2 seconds with 71 samples under non SDK mode. 

I would like to know why Verity Sense ACC stream comes every second or 2 seconds and not consistently comes every second .
And also would like to know why the y value has negative value when the Verity Sense sensor button is in up position. 
Why x y z values of Verity Sense are so different from OH1.

The following is the ACC stream code for OH1 and Verity Sense.

            var accStreamSettings: StreamSettings?
            if appdelegate.sensorName == "OH1"{
                accStreamSettings =
            StreamSettings( feature: DeviceStreamingFeature.acc,
                            settings:[ StreamSetting(type: PolarSensorSetting.SettingType.sampleRate, values:[50]),
                                       StreamSetting(type: PolarSensorSetting.SettingType.range, values: [8]),
                                       StreamSetting(type: PolarSensorSetting.SettingType.resolution, values: [16])])
            }
            else if appdelegate.sensorName == "Sense"{
                accStreamSettings =
                StreamSettings( feature: DeviceStreamingFeature.acc,
                                settings:[ StreamSetting(type: PolarSensorSetting.SettingType.sampleRate, values:[52]),
                                           StreamSetting(type: PolarSensorSetting.SettingType.range, values: [8]),
                                           StreamSetting(type: PolarSensorSetting.SettingType.resolution, values: [16]),
                                           StreamSetting(type: PolarSensorSetting.SettingType.channels, values: [3])])
                
            }

     accToggle = api.startAccStreaming(appdelegate.sensorId, settings: PolarSensorSetting(polarSensorSettings)).observe(on:MainScheduler.instance).subscribe{ e in
                
                
                switch e { 
                 case .next(let data):   
                 ~  
                 }
***Verity Sense ACC Stream datas *****
acc timeStamp 2022/11/21 10:54:22.817) samplesCount 71 data (timeStamp: 722343271643846165, samples: [(x: 35, y: -980, z: -62), (x: 41, y: -982, z: -65), (x: 40, y: -983, z: -76), ~~      Omitted   ~~  ,  (x: 39, y: -978, z: -51), (x: 48, y: -977, z: -69), (x: 44, y: -981, z: -76), (x: 33, y: -982, z: -70)])　

acc timeStamp 2022/11/21 10:54:23.817) samplesCount 71 data (timeStamp: 722343272959519870, samples: [(x: 37, y: -978, z: -69), (x: 38, y: -978, z: -72), (x: 27, y: -981, z: -70),  ~~      Omitted   ~~   , (x: 44, y: -965, z: -60)])　

acc timeStamp 2022/11/21 10:54:25.817) samplesCount 71 data (timeStamp: 722343274275246977, samples: [(x: 40, y: -965, z: -62), (x: 41, y: -969, z: -64), (x: 32, y: -968, z: -75), ~~      Omitted   ~~   , (x: 23, y: -976, z: -73), (x: 30, y: -980, z: -74), (x: 22, y: -977, z: -72)])　

***OH1 ACC Stream datas *****
acc timeStamp 2022/11/21 11:03:25.575) samplesCount 36 data (timeStamp: 540815406492180684, samples: [(x: -211, y: 940, z: 248), (x: -212, y: 938, z: 250), (x: -213, y: 939, z: 250), (x: -216, y: 937, z: 252), (x: -216, y: 939, z: 251),  ~~      Omitted   ~~   , (x: -211, y: 938, z: 252), (x: -210, y: 940, z: 251), (x: -210, y: 942, z: 253)])　

acc timeStamp 2022/11/21 11:03:26.575) samplesCount 36 data (timeStamp: 540815407203795966, samples: [(x: -209, y: 942, z: 252), (x: -209, y: 945, z: 253), (x: -210, y: 941, z: 253), (x: -209, y: 939, z: 251),  ~~      Omitted   ~~    , (x: -210, y: 942, z: 257), (x: -209, y: 941, z: 258)])

JOikarinen, 2022-11-21:
Hi @ymatt345, thanks for the question. 

> ACC stream datas from OH1 come every second with 36 samples.
> However ACC stream datas from VeritySense comes every second or 2 seconds with 71 samples under non SDK mode.

Both of the devices are using same size data transmission packets (232 bytes) when data is sent over BLE. The device buffers the data until it has enough data to fill the transmission packet. However, the OH1 is not compressing the data when filling the transmission packet, whereas VeritySense does the compression. The result is that VeritySense needs to sent the data less frequently. Furthermore, the VeritySense compression is based on changes in consecutive samples, if the device is not moving much the compression is more effective. Thus the time between received transmission packets from VeritySense may vary.

JOikarinen, 2022-11-21:
> And also would like to know why the y value has negative value when the Verity Sense sensor button is in up position. Why x y z values of Verity Sense are so different from OH1.

The axle directions are different in VS and OH1. The X direction is same between VS and OH1. But the Y and Z axles are opposite directions. Here is the axle direction map of VeritySense:

[image]

ymatt345, 2022-11-21:
I could clearly understand the all my concerns.
I can now modify our ACC streaming analysis logic.
Thank you for your prompt response.
ymatt345.

closed 2022-11-21 by ymatt345