polarofficial/polar-ble-sdk #318: What is the Sample Rate of Verity Sense PPG Streaming

ymatt345, 2022-11-21:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [x] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

I am working on the additional  support of Verity Sense to my App that already supports OH1.
I compared the PPG streaming datas from VeritySense with the one from OH1.

PPG stream datas from OH1 comes every 133 mil-sec with 18 samples that is close to SampleRate value 130 specified by the StreamSettings. 
PPG stream datas from VeritySense comes every 761.9 mil-sec with 42 samples that is very different from the  SampleRate value 135 specified by the StreamSettings under non SDK mode.

So I would like to know the definition of Sample Rate for PPG Streaming.
And also would like to know the way to set the interval time for PPG streaming datas.

The following is the ACC stream code for OH1 and Verity Sense.

            var ppgStreamSettings: StreamSettings?
            if appdelegate.sensorName == "OH1"{
            ppgStreamSettings =
            StreamSettings( feature: DeviceStreamingFeature.ppg,
                            settings:[ StreamSetting(type: PolarSensorSetting.SettingType.sampleRate, values:[130]),
                                       StreamSetting(type: PolarSensorSetting.SettingType.resolution, values: [22])])
            }
            else if appdelegate.sensorName == "Sense"{
               ppgStreamSettings =
                StreamSettings( feature: DeviceStreamingFeature.ppg,
                                settings:[ StreamSetting(type: PolarSensorSetting.SettingType.sampleRate, values:[135]),
                                           StreamSetting(type: PolarSensorSetting.SettingType.resolution, values: [22]),
                                           StreamSetting(type: PolarSensorSetting.SettingType.channels, values: [4])])
            }
            ppgToggle = api.startOhrStreaming(self.appdelegate.sensorId, settings: PolarSensorSetting(polarSensorSettings)).observe(on: MainScheduler.instance).subscribe{ [self] e in
                switch e {
                case .completed:
                    NSLog("ppg finished")
                case .error(let err):
                    NSLog("start ppg error: \(err)")
                    self.ppgToggle = nil
                case .next(let data):
                  ~   Omitted ~
               }
 
Verity Sense PPG stream datas
timeStamp 2022/11/21 11:27:14.958 bucketTimeDiff 761.9063720703125 loopCount 0 samples.count 42 item [444333, 556166, 615661, 185081]
timeStamp 2022/11/21 11:27:14.958 bucketTimeDiff 761.9063720703125 loopCount 1 samples.count 42 item [444461, 556285, 615772, 185126]
timeStamp 2022/11/21 11:27:14.958 bucketTimeDiff 761.9063720703125 loopCount 2 samples.count 42 item [444487, 556293, 615828, 185106]
~   Omitted ~
timeStamp 2022/11/21 11:27:14.958 bucketTimeDiff 761.9063720703125 loopCount 40 samples.count 42 item [445898, 558421, 617715, 185061]
timeStamp 2022/11/21 11:27:14.958 bucketTimeDiff 761.9063720703125 loopCount 41 samples.count 42 item [445879, 558333, 617618, 185070]

OH1  PPG stream datas
timeStamp 2022/11/21 11:03:25.635 bucketTimeDiff 133.05450439453125 loopCount 0 samples.count 18 item [-139002, -31630, -8082, -326723]
timeStamp 2022/11/21 11:03:25.635 bucketTimeDiff 133.05450439453125 loopCount 1 samples.count 18 item [-138869, -31491, -8248, -326596]
timeStamp 2022/11/21 11:03:25.635 bucketTimeDiff 133.05450439453125 loopCount 2 samples.count 18 item [-138886, -31406, -8201, -326549]
~   Omitted ~
timeStamp 2022/11/21 11:03:25.635 bucketTimeDiff 133.05450439453125 loopCount 16 samples.count 18 item [-138611, -31011, -7903, -326565]
timeStamp 2022/11/21 11:03:25.635 bucketTimeDiff 133.05450439453125 loopCount 17 samples.count 18 item [-138560, -30888, -7805, -326540]

JOikarinen, 2022-11-22:
Hi @ymatt345,

sample rate is the rate the sampling is done in the device, but not the rate the data is delivered over bluetooth. As explained in #317 the VeritySense is compressing data before sending it over BLE. If comparing the OH1 and VS, as you observed, the data from VS arrives less frequently. It means data is efficiently packed and not need to be delivered that frequently. 

However, there is [a bug](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/KnownIssues.md#issue-1) related to PPG sampling rate. The sampling rate is not the promised 135Hz, but 55Hz.

ymatt345, 2022-11-24:
Thank you for your comments.
It requires complicated techniques to assign calculated time stamp on each data samples compared to OH1.
Is possible to assign 28Hz sampling rate for PPG under SDK mode?
If so I will do so.

JOikarinen, 2022-12-13:
Hi @ymatt345, sorry for late response. 

> It requires complicated techniques to assign calculated time stamp on each data samples compared to OH1.

you may take a look the new [release 4.0.0-beta0](https://github.com/polarofficial/polar-ble-sdk/releases/tag/4.0.0-beta0) where the time stamp is given for every sample

> Is possible to assign 28Hz sampling rate for PPG under SDK mode?

Yes it is. https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/SdkModeExplained.md

closed 2022-12-13 by JOikarinen