polarofficial/polar-ble-sdk #325: Polar H10 internal calculations

RaySub, 2023-01-30:
***Platform your question concerns:*** 
- [X ] Android through the use of Polar Sensor Logger
- [ ] iOS
- [ ] Other
- [ X] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [X ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

1. I would like to align the detected qrs peaks with the ECG record
- What does the timestamp of an RR interval stands for: beginning, centre or end of an RR interval? Anything else?

2. How is the parameter ppErrorEstimate being calculated? By aggregating on sliding time windows such as when calculating rolling (moving) averages?

3. As noted in issue [#13](https://github.com/polarofficial/polar-ble-sdk/issues/13) a heart rate measurement in beats per minute (± 2 to 5%) is stored once every second which is suboptimal. How is this measurement being estimated? I noticed a severe overestimation of HR where important variations exist between consecutive RR intervals. This should not be the case since RR intervals are accurately estimated. Indeed I could visually check on an ECG trace that the H10' qrs detector performed well in such cases.

RaySub, 2023-01-30:
In the attached txt file, you may count 51 qrs in 23 seconds. Mean + sd HR should be 133.04 ± 25.7 bpm in place of 170 – 174 bpm. Note that 170 – 174 also corresponds to the range of values that were displayed in realtime on the bike computer paired to the H10.

[dtt.txt](https://github.com/polarofficial/polar-ble-sdk/files/10536755/dtt.txt)

andYgnite, 2025-07-22:
Hi @liegepr 

RR interval timestamp is from the end of the interval.

I can't unfortunately comment on other 2 questions.

closed 2025-07-22 by andYgnite