polarofficial/polar-ble-sdk #326: [CoreBluetooth] API MISUSE: <CBCentralManager: 0x281a001e0> can only accept this command while in the powered on state

FeherF, 2023-01-31:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [x] Device is not relevant for this question

I get the following error while I initialize the `api` variable:

**[CoreBluetooth] API MISUSE: <CBCentralManager: 0x281a001e0> can only accept this command while in the powered on state**

Here's the interested block of code:
```
@objc class PolarController: NSObject,
                    PolarBleApiObserver,
                    PolarBleApiPowerStateObserver,
                    PolarBleApiDeviceInfoObserver,
                    PolarBleApiDeviceFeaturesObserver,
                    PolarBleApiDeviceHrObserver {
    
    // Error on this line
    var api = PolarBleApiDefaultImpl.polarImplementation(DispatchQueue.main, features: Features.allFeatures.rawValue)

    @objc func initAndConnect() {
        api.polarFilter(false)
        api.observer = self
        api.deviceHrObserver = self
        api.powerStateObserver = self
        api.deviceFeaturesObserver = self
        api.deviceInfoObserver = self
        do {
            try self.api.connectToDevice("deviceID")
        } catch {
            print("Error occurred while connecting to device: \(error)")
        }
    }
```

I'm testing on iPhone 11 with iOS 16, I'm using polar-ble-sdk 3.3.6v.
Thanks in advance!

JOikarinen, 2023-02-09:
Hi @FeherF, 

you probably see the issue when launching your app using Xcode? The logs appear if relaunching the app and there was connection at the time of relaunch of the app. The origin of the problem is the Bluetooth State Preservation and Restoration, which was introduced in release 3.3.4. The Bluetooth State Preservation and Restoration will be removed in release 5.0.0 as it is not working properly.

closed 2023-02-09 by JOikarinen

JOikarinen, 2023-02-13:
The Bluetooth State Preservation and Restoration is now removed in release [5.0.0-beta2](https://github.com/polarofficial/polar-ble-sdk/releases/tag/5.0.0-beta2)

FeherF, 2023-02-13:
Thank you very much!