polarofficial/polar-ble-sdk #330: Polar devices not found when using searchForDevice() method in determined terminals on Android 10

kitkatbonsai, 2023-02-16:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [x] Device is not relevant for this bug

**Describe the bug**
In determined smartphones and tablets with Android 10, POLAR devices are visible and pairable from settings but not visible after calling PolarApi.searchForDevice().

**How to Reproduce**
Just open from VCS the example located in https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp and try to scan devices with a Lenovo Tab M10 HD or some Xiaomi devices... also tried with the provided Demo in https://github.com/polarofficial/polar-ble-sdk/tree/master/demos/Android-Demos/PolarSDK-ECG-HR-Demo.

**Expected behavior**
POLAR devices perfectly visible from bluetooth settings, but the app is not able to find any device.

**Encountered solution**
After installing Polar Flow to check if it had something to do with the PolarBleAPI, I have noticed that the examples and demos app of PolarBleSDK do not require ACCESS_BACKGROUND_LOCATION whereas Polar FLow does.

I have implemented the permission in the case of Android 10 and below and now it works perfectly nice while scanning devices again. Maybe this information could be added in "known issues" or somewhere else in the intro of the PolarBleSdk Github page as a warning or piece of advice in case of the api not scanning as expected.

JOikarinen, 2023-03-13:
Hi @kitkatbonsai, 

thanks for the detailed description. I double check your observation with Nokia 9 having Android 10, I couldn't reproduce. 

Only case the ACCESS_BACKGROUND_LOCATION permission is needed by Polar BLE SDK when Polar BLE SDK is used in service which is not declared as foreground service. That statement is based on the [startScan documentation](https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner#startScan(android.bluetooth.le.ScanCallback)) and [Bluetooth permission documentation](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions)

 So it sounds like your observation is Android device specific. Before I will proceed I could also play around with Xiaomi having the Android 10.

JOikarinen, 2023-03-15:
Btw, now that I tested the issue described with Xiaomi Note 7 (Android 10) I cannot reproduce issue. Example app and demo app works okay without ACCESS_BACKGROUND_LOCATION

kitkatbonsai, 2023-04-11:
Hi @JOikarinen, thank you very much for your efforts and apologies for my delayed response.

I did have problems with two specific devices with Android 10 (having installed the demo apps and my app as well): Xiaomi A2, Lenovo TAB M10. Maybe it is just an Android device specific issue, as you well indicated.

closed 2023-09-08 by jimmyzumthurm