polarofficial/polar-ble-sdk #331: Encountered frame type 129

ghup1, 2023-02-17:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
An unknown frame type 129 came up after starting a stream for accelerometer data. This frame type is not documented, and I cannot fully understand the format.

**Start ACC stream request:**
`0x02 0x02 0x00 0x01 0x34 0x00 0x01 0x01 0x10 0x00 0x02 0x01 0x08 0x00 0x04 0x01 0x03`

Resolution: 0x10 = 16 bits
Channels: 0x03 = 3 channels

**Control point response:**
`0xf0 0x02 0x02 0x00 0x00`

Error code: 0x00 = Stream successfully started

**First data:**
`0x2 0x9c 0x58 0x29 0x13 0x84 0x57 0x73 0x7 0x81 0xfa 0xff 0x21 0x0 0xf4 0x3 0x6 0x3b 0x83 0xd0 0xeb 0xc6 0xf1 0xef 0x3a 0xf2 0x1b 0xfd 0xef 0x7 0xbf 0x1f 0x0 0x77 0xf 0x24 0x7e 0x30 0x10 0x41 0xef 0xf 0x3f 0xe0 0xb 0xff 0xef 0xf 0x42 0xe1 0x3 0x80 0x0 0xe0 0xc2 0x50 0xf4 0xbd 0xf 0x0 0x3 0xe0 0xf7 0x7e 0x40 0x8 0x42 0x2f 0x0 0xc0 0xaf 0x17 0x3b 0xa1 0x27 0xc0 0xb0 0x3 0x3d 0x30 0xf4 0x7c 0x6e 0x10 0x83 0xfe 0x3 0xbf 0xe0 0x1b 0x43 0x81 0xf3 0x3e 0x22 0x0 0x3c 0x30 0x0 0x3e 0x20 0x10 0x7e 0xcf 0xff 0xca 0xef 0xe3 0x81 0x60 0x0 0x3f 0x3f 0x0 0x5 0xf 0xe8 0xc2 0xf 0xc 0xbe 0xdf 0x1f 0x80 0xef 0x13 0x4 0x7f 0xf7 0x80 0x10 0x8 0x9 0xf0 0xf7 0x3f 0xb0 0x13 0x4 0xd2 0xff 0x3d 0xf0 0xf3 0xc2 0x4f 0xf4 0xc0 0xff 0xff 0x1 0x4 0xa 0x34 0xd2 0x2e 0xf3 0xbc 0xf2 0xf2 0x4e 0x5b 0x41 0xc0 0xe0 0x5f 0xff 0xc 0x6 0xa 0x82 0x30 0xf8 0xfe 0x10 0x10 0xbb 0xf 0x18 0x3 0x3f 0xff 0x2 0x12 0xfc 0x7a 0x10 0x0 0xbd 0x6e 0x4 0x7 0x0 0x4 0xa 0x2 0xa1 0x2e 0xf4 0x4f 0xff 0xff 0xbf 0xf4 0xd2 0x3f 0x34 0xf0 0xe2 0xef 0x6 0x7 0x82 0x30 0xf8 0x41 0xfe 0xdf 0x82 0x60 0xe8 0xc1 0x6f 0xf0 0x3b 0x0 0x18 0x0`

Measurement type: 0x02 = accelerometer
Timestamp 0x9c 0x58 0x29 0x13 0x84 0x57 0x73 0x7
Frame type: 0x81 = 129
Delta size: 0x04 = 4 bits

Especially the frame type and delta size are not expected. Am I missing something, or is the documentation not up-to-date?

ghup1, 2023-02-20:
So when I treat the frame as a delta frame, it works fine.

closed 2023-02-20 by ghup1

fagerbur, 2023-04-24:
@ghup1 I came across the same frame type but I'm not sure how to decode the timestamps that are now included in the individual delta samples based on the latest release, my existing translation script reports numbers out of range when i treat it as a delta frame so I'm trying to convert to the new method. Do you know if there is a timestamp associated with the reference sample still? How many bytes are you allowing for the timestamp within each delta sample?