polarofficial/polar-ble-sdk #338: Incomplete aar build.

muammar-yacoob, 2023-02-23:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Incomplete library files
Hi! Could you please provide the `aar` file to use in an android project? I keep getting these errors on my build which is clearly missing required files
`AndroidJavaException: java.lang.NoClassDefFoundError: Failed resolution of: Lkotlin/jvm/internal/Intrinsics;`

Thank you!

JOikarinen, 2023-03-15:
Hi @muammar-yacoob, sorry for late reply. Are you still having the issue? How do you make dependency to Polar BLE SDK? Are you following these steps: https://github.com/polarofficial/polar-ble-sdk#installation

jimmyzumthurm, 2023-09-08:
Hi @muammar-yacoob  , I will close this issue as you did not give any update. Please feel free to re-open if you still have issues.

closed 2023-09-08 by jimmyzumthurm