polarofficial/polar-ble-sdk #343: H10: RR Interval in ms and 1/1024 format.

Retsam-crypto, 2023-03-05:
***Device:***
- [ ] Polar H10

What the **1/1024 format** means for RRI, please (found ion first page of SDK documentation)?

Silent assumption is that sensor sampling rate is 1000 Hz, RR peaks are discovered from ECG internally, then ECG is down sampled to 130 Hz. But that assumption may be incorrect and sampling rate internally is 1024 Hz. (hence the 1/1024 'format' mentioned in SDK documentation).

Please shed some light on that process.

JOikarinen, 2023-03-16:
Hi @Retsam-crypto, 

The format of 1/1024 to represent RR time has been a source of confusion. RRI is the time between the two consecutive R waves in the electrocardiogram (ECG) signal.

The format 1/1024 to represent the RRI time is quite strange. [Heart Rate BLE service](https://www.bluetooth.com/specifications/specs/heart-rate-service-1-0/) provides the RR data for HR BLE client in 1/1024 format. I suspect that reason for HR service using the 1/1024 format is the limited resources  in some heart rate sensor devices at the time specification was made (approximately 10 years ago). 

Why the Polar BLE SDK API exposes the 1/1024 format? Well, it is provided in API to let raw data pass through from heart sensor without any manipulation by the Polar BLE SDK. 

However, due to the confusion, the Polar BLE SDK API will be updated in the next release (5.0.0) to only use the "milliseconds" format for RR time. The `PolarHrData` will have only one property, rrs, instead of rrs and rrsMs. The Android and iOS changes can be viewed in the provided links.

Change in Android: [here](https://github.com/polarofficial/polar-ble-sdk/commit/011d7f1c5dda2b4b929be82f678685cdf8cb5a81#diff-fddf4aa48d91c713b39e6e901e2e0f14fcd09a8c327ef52260ebdc29969ff80f)

Change in iOS: [here](https://github.com/polarofficial/polar-ble-sdk/commit/011d7f1c5dda2b4b929be82f678685cdf8cb5a81#diff-2557a7ddfd5ef4d1fdaa16de7d250a80e87872e6766f41ff7b147a2e2e2ff0bd)

JOikarinen, 2023-03-16:
It was not a good decision to rename `rrs` to `rrsMs` and remove `rrsMs`, done in Android: [here](https://github.com/polarofficial/polar-ble-sdk/commit/011d7f1c5dda2b4b929be82f678685cdf8cb5a81#diff-fddf4aa48d91c713b39e6e901e2e0f14fcd09a8c327ef52260ebdc29969ff80f) iOS: [here](https://github.com/polarofficial/polar-ble-sdk/commit/011d7f1c5dda2b4b929be82f678685cdf8cb5a81#diff-2557a7ddfd5ef4d1fdaa16de7d250a80e87872e6766f41ff7b147a2e2e2ff0bd)

Better idea implemented, let's remove `rrs` (i.e. format 1/1024) and let's leave only the property `rrsMs`, done in Android: [here](https://github.com/polarofficial/polar-ble-sdk/commit/76437f0b9713bb2e423f40cdf418e4bbb05c66c7#diff-fddf4aa48d91c713b39e6e901e2e0f14fcd09a8c327ef52260ebdc29969ff80f) iOS: [here](https://github.com/polarofficial/polar-ble-sdk/commit/76437f0b9713bb2e423f40cdf418e4bbb05c66c7#diff-2557a7ddfd5ef4d1fdaa16de7d250a80e87872e6766f41ff7b147a2e2e2ff0bd)

closed 2023-03-16 by JOikarinen