polarofficial/polar-ble-sdk #344: Some BLE exchange protocol?

hheimerd, 2023-03-07:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

Can you provide a document in which the actions will be fully described for connecting to the device and setting it up for obtaining PPG data? I’ve been trying for several hours to find bit from other people in previous Issues. I was able to get AAC data, but not PPG, (I could not get this data before connecting the device to the phone). I inserted a configuration of #226, but I do not know how to change the type of frame from Delta Frame to X, Y, Z. I don't even know what Delta Frame is.

`[0x02,
    0x02,
    0x00,
    0x01,
    0x34,
    0x00,
    0x01,
    0x01,
    0x10,
    0x00,
    0x02,
    0x01,
    0x08,
    0x00,
    0x04,
    0x01,
    0x03
]`

Ideally, I would like a document where every byte is described, like here:
[image]

andYgnite, 2023-07-10:
Hi @hheimerd , 
could you please gives us more details?

You can directly use SDK to stream ACC data from Polar sensors.
Delta frame is the data compressor.

hheimerd, 2023-07-10:
I found one repository on JS and found in it how to interpret the data. Maybe it will help someone.
[parser.zip](https://github.com/polarofficial/polar-ble-sdk/files/12001115/parser.zip)
It is necessary to describe all these transformations in a good way so that they can be implemented in other languages and contexts. For example, I needed to make an application for a PC

andYgnite, 2023-07-11:
Hi @hheimerd, good you could find the parser.
Thank you for your suggestion; we will definitely take it into consideration.

closed 2023-10-12 by samulimaa