polarofficial/polar-ble-sdk #346: Android dependency resolution issue

Rexios80, 2023-03-13:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [x] Device is not relevant for this bug

**Describe the bug**
Another of our dependencies depends on https://github.com/dlew/joda-time-android which is incompatible with joda-time. Accoring to https://github.com/dlew/joda-time-android/issues/184#issuecomment-312645015 you shouldn't use the raw joda-time dependency on Android, so maybe we should replace it?

**How to Reproduce**
Include a dependency that depends on joda-time-android

**Expected behavior**
No dependency resolution errors

**Screenshots and logs**
If applicable, add screenshots or logs to help explain your problem.

```console
Duplicate class org.joda.time.Chronology found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateMidnight found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateMidnight$Property found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTime found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTime$Property found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeComparator found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeConstants found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeField found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeFieldType found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeFieldType$StandardDateTimeFieldType found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeUtils found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeUtils$FixedMillisProvider found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeUtils$MillisProvider found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeUtils$OffsetMillisProvider found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeUtils$SystemMillisProvider found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeZone found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeZone$LazyInit found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeZone$LazyInit$1 found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DateTimeZone$Stub found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.Days found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.Duration found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DurationField found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DurationFieldType found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.DurationFieldType$StandardDurationFieldType found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.Hours found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.IllegalFieldValueException found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.IllegalInstantException found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.Instant found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.Interval found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.JodaTimePermission found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.LocalDate found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.LocalDate$Property found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.LocalDateTime found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.LocalDateTime$Property found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.LocalTime found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.LocalTime$Property found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.Minutes found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.MonthDay found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.MonthDay$Property found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
[... cut ...]
Duplicate class org.joda.time.tz.ZoneInfoLogger$1 found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.tz.ZoneInfoProvider found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)
Duplicate class org.joda.time.tz.ZoneInfoProvider$1 found in modules jetified-joda-time-2.12-no-tzdb.1-no-tzdb (joda-time:joda-time:2.12.1) and jetified-joda-time-2.12.1 (joda-time:joda-time:2.12.1)

Go to the documentation to learn how to Fix dependency resolution errors.
```

Rexios80, 2023-03-16:
Do you have an ETA for that?

JOikarinen, 2023-03-17:
> Do you have an ETA for that?

@Rexios80 unfortunately no

Rexios80, 2023-03-17:
Since you are in the process of making a new major version, would it make sense to include that change?

If so, can I create a PR for that?

Also, how do I open the project in Android Studio? I can't get the sdk folder to show up in the folder tree, and no dependencies resolve in those files. It seems like the sdk files should be in a separate module but they aren't.

lilmikey243, 2023-04-23:
Duplicate class android.support.v4.app.INotificationSideChannel found in modules core-1.9.0-runtime (androidx.core:core:1.9.0) and support-compat-27.1.1-runtime (com.android.support:support-compat:27.1.1)
Duplicate class android.support.v4.app.INotificationSideChannel$Stub found in modules core-1.9.0-runtime (androidx.core:core:1.9.0) and support-compat-27.1.1-runtime (com.android.support:support-compat:27.1.1)
Duplicate class android.support.v4.app.INotificationSideChannel$Stub$Proxy found in modules core-1.9.0-runtime (androidx.core:core:1.9.0) and support-compat-27.1.1-runtime (com.android.support:support-compat:27.1.1)
Duplicate class android.support.v4.os.IResultReceiver found in modules core-1.9.0-runtime (androidx.core:core:1.9.0) and support-compat-27.1.1-runtime (com.android.support:support-compat:27.1.1)
Duplicate class android.support.v4.os.IResultReceiver$Stub found in modules core-1.9.0-runtime (androidx.core:core:1.9.0) and support-compat-27.1.1-runtime (com.android.support:support-compat:27.1.1)
Duplicate class android.support.v4.os.IResultReceiver$Stub$Proxy found in modules core-1.9.0-runtime (androidx.core:core:1.9.0) and support-compat-27.1.1-runtime (com.android.support:support-compat:27.1.1)
Duplicate class android.support.v4.os.ResultReceiver found in modules core-1.9.0-runtime (androidx.core:core:1.9.0) and support-compat-27.1.1-runtime (com.android.support:support-compat:27.1.1)
Duplicate class android.support.v4.os.ResultReceiver$1 found in modules core-1.9.0-runtime (androidx.core:core:1.9.0) and support-compat-27.1.1-runtime (com.android.support:support-compat:27.1.1)
Duplicate class android.support.v4.os.ResultReceiver$MyResultReceiver found in modules core-1.9.0-runtime (androidx.core:core:1.9.0) and support-compat-27.1.1-runtime (com.android.support:support-compat:27.1.1)
Duplicate class android.support.v4.os.ResultReceiver$MyRunnable found in modules core-1.9.0-runtime (androidx.core:core:1.9.0) and support-compat-27.1.1-runtime (com.android.support:support-compat:27.1.1)

Go to the documentation to learn how to Fix dependency resolution errors.

Rexios80, 2023-04-26:
That doesn't look like the same issue

closed 2026-08-28 by Rexios80