polarofficial/polar-ble-sdk #350: H10 disconnecting on some devices

ThomasReyskens, 2023-03-16:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Device disconnects immediately after connecting.
I use the same code on 10 phones. 5 of them work perfectly, on the other ones, the polar devices cannot stay connected.

**How to Reproduce**
Connect to the polar device, wait for 2 secs -> device disconnects

**Expected behavior**
device stays connected

**Screenshots and logs**
logs: 

```
2023-03-16 14:25:56.099 18372-18372 DashboardPresenter               E  Going to connect to:C2920821

2023-03-16 14:25:56.108 18372-18372 POLAAAR                          E  BDScanCallback/commandState state:IDLE action: CLIENT_START_SCAN
2023-03-16 14:25:56.108 18372-18372 POLAAAR                          E  BDScanCallback/commandState state:IDLE action: EXIT
2023-03-16 14:25:56.108 18372-18372 POLAAAR                          E  BDScanCallback/commandState state:SCANNING action: ENTRY
2023-03-16 14:25:56.108 18372-18372 POLAAAR                          E  BDScanCallback/timestamps left: 0
2023-03-16 14:25:56.108 18372-18372 POLAAAR                          E  BDScanCallback/Scan started -->
2023-03-16 14:25:56.108 18372-18372 BluetoothAdapter                 D  isLeEnabled(): ON
2023-03-16 14:25:56.110 18372-19618 BluetoothLeScanner               D  onScannerRegistered() - status=0 scannerId=18 mScannerId=0
2023-03-16 14:25:56.118 18372-18372 POLAAAR                          E  BDScanCallback/Scan started <--

023-03-16 14:25:57.084 18372-18372 POLAAAR                          E  BleGattBase/Added notification characteristic for fb005c81-02e7-f387-1cad-8acd2d8df0c8
2023-03-16 14:25:57.085 18372-18372 POLAAAR                          E  BleGattBase/Added notification characteristic for fb005c82-02e7-f387-1cad-8acd2d8df0c8
2023-03-16 14:25:57.086 18372-18372 POLAAAR                          E  BleGattBase/Added notification characteristic for fb005c51-02e7-f387-1cad-8acd2d8df0c8
2023-03-16 14:25:57.086 18372-18372 POLAAAR                          E  BleGattBase/Added notification characteristic for fb005c52-02e7-f387-1cad-8acd2d8df0c8
2023-03-16 14:25:57.087 18372-18372 POLAAAR                          E  BleGattBase/Added notification characteristic for 00002a19-0000-1000-8000-00805f9b34fb
2023-03-16 14:25:57.087 18372-18372 POLAAAR                          E  BDDeviceListenerImpl/new device allocated name: Polar H10 C2920821
2023-03-16 14:25:57.087 18372-18372 POLAAAR                          E  BDDeviceList/new session added: Polar H10 C2920821
2023-03-16 14:25:57.088 18372-18372 POLAAAR                          E  BDScanCallback/commandState state:SCANNING action: CLIENT_REMOVED
2023-03-16 14:25:57.088 18372-18372 POLAAAR                          E  BDScanCallback/commandState state:SCANNING action: EXIT
2023-03-16 14:25:57.088 18372-18372 POLAAAR                          E  BDScanCallback/Stop scanning
2023-03-16 14:25:57.088 18372-18372 BluetoothAdapter                 D  isLeEnabled(): ON
2023-03-16 14:25:57.088 18372-19688 POLAAAR                          E  ConnectionHandler/state: CONNECTING action: ENTRY
2023-03-16 14:25:57.088 18372-19688 POLAAAR                          E  BDScanCallback/commandState state:SCANNING action: ADMIN_STOP_SCAN
2023-03-16 14:25:57.088 18372-19688 POLAAAR                          E  BDScanCallback/commandState state:SCANNING action: EXIT
2023-03-16 14:25:57.088 18372-19688 POLAAAR                          E  BDScanCallback/Stop scanning
2023-03-16 14:25:57.088 18372-19688 BluetoothAdapter                 D  isLeEnabled(): ON
2023-03-16 14:25:57.100 18372-19688 BluetoothLeScanner               D  could not find callback wrapper
2023-03-16 14:25:57.100 18372-18372 POLAAAR                          E  BDScanCallback/commandState state:IDLE action: ENTRY
2023-03-16 14:25:57.100 18372-19688 POLAAAR                          E  BDScanCallback/commandState state:STOPPED action: ENTRY
2023-03-16 14:25:57.100 18372-19688 POLAAAR                          E  ConnectionHandler/ Session update from: SESSION_CLOSED to: SESSION_OPENING
2023-03-16 14:25:57.100 18372-19688 POLAAAR                          E  BDScanCallback/commandState state:STOPPED action: CLIENT_REMOVED
2023-03-16 14:25:57.111 18372-18372 DashboardPresenter               E  DEVICE CONNECTING
2023-03-16 14:25:57.113 18372-19688 BluetoothGatt                    D  connect() - device: D7:D5:0F:3E:3D:55, auto: false, eattSupport: false
2023-03-16 14:25:57.113 18372-19688 BluetoothGatt                    D  registerApp()
2023-03-16 14:25:57.113 18372-19688 BluetoothGatt                    D  registerApp() - UUID=a635b87c-8da3-4658-9d87-0792652606de
2023-03-16 14:25:57.117 18372-19688 POLAAAR                          E  connect search completed for C2920821
2023-03-16 14:25:57.117 18372-19564 BluetoothGatt                    D  onClientRegistered() - status=0 clientIf=18

2023-03-16 14:25:59.355 18372-19564 BluetoothGatt                    D  onClientConnectionState() - status=0 clientIf=18 device=D7:D5:0F:3E:3D:55
2023-03-16 14:25:59.357 18372-19564 POLAAAR                          E  GattCallback/GATT state changed device newState: 2 status: 0
2023-03-16 14:25:59.609 18372-19564 BluetoothGatt                    D  onConnectionUpdated() - Device=D7:D5:0F:3E:3D:55 interval=6 latency=0 timeout=500 status=0
[... 30 lines cut ...]
```

All permission are set via ``` if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.Q) {
      if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.S) {
        permissions = new String[] {
                Manifest.permission.BLUETOOTH_SCAN,
                Manifest.permission.BLUETOOTH_CONNECT,
                Manifest.permission.ACCESS_FINE_LOCATION,
                Manifest.permission.ACCESS_COARSE_LOCATION,
                Manifest.permission.CAMERA
        };
      } else {
        permissions = new String[] {
                Manifest.permission.ACCESS_FINE_LOCATION,
                Manifest.permission.ACCESS_COARSE_LOCATION,
                Manifest.permission.CAMERA
        };
      }
    } else {
      permissions = new String[] {
                Manifest.permission.ACCESS_FINE_LOCATION,
                Manifest.permission.ACCESS_COARSE_LOCATION,
                Manifest.permission.CAMERA
      };
    }```

JOikarinen, 2023-03-17:
Hi @ThomasReyskens 

May I assume that the phones you have observed the issue are having Android 13?

Here is the official statement related to H9 and H10 and Android13 compatibility issues:

"We are sorry to hear about this trouble with your heart rate sensor. The issue seems to be caused by the recent Android 13 update, and we are aware that the problem has been fixed already to certain phones like Samsung and Google Pixel, so please check first if there's a newer Android version available for your phone. It might solve the issue you’re experiencing.
Despite of this, our team at Polar has been working on a solution for this issue as well to ensure all our customers get their sensors back to work with the Android 13, and therefore we will release a new firmware version for our heart rate sensors. The date is yet to be confirmed, but you will be notified about the update availability via Polar Flow and Beat apps and at the Updates at support.polar.com."

ThomasReyskens, 2023-03-17:
Hi @JOikarinen!

Almost all the devices where it doesn't work are indeed on Android 13. There is one: Pocophone F1 (Android 10) that also disconnects.
But good to know that you are working on a fix :) 
Thanks for the clear explanation, looking forward to the update!

closed 2023-03-17 by ThomasReyskens

kitkatbonsai, 2023-03-27:
Hello there, this also happened to me with OPPO A54 and OnePlus 5T, apparently, also with Android 13 installed. However, on samsung mobiles with Android 13 this behavior is not reproduced. Looks like a issue with most chinese phones, namely with BBK Electronics-based phones (Realme, OnePlus, OPPO...). Maybe caused by their personalized Android systems (Color OS, MIUI, etc...)?

Hope the new version of the SDK is soon released. Thanks for the good work on it!