polarofficial/polar-ble-sdk #351: Accelerometer raw data

mebnali, 2023-03-21:
Dear polar team,
We need to export Accelerometer  data. Is there any option for us to do?
I appreciate your help.

JOikarinen, 2023-03-23:
Hi @mebnali, could you explain more detailed? This SDK here is helping you to stream ACC data from Polar sensors.

mebnali, 2023-04-08:
Thank you. What i see in the raw data, it only shows one column for acceleration data. I was expecting acceleration in three directions (x, y , z). What is the didfference here? Is there anyway for us to get acceerlaiton data in three directions?

andYgnite, 2023-07-10:
Hi @mebnali, using the SDK you should be able to get X,Y,Z acceleration data. Can you provide more details about steps you follow to get only one column of data?

jimmyzumthurm, 2023-09-08:
Hi @mebnali , I'm closing this now as we didn't get enough details about your issue to help, but feel free to  make a new ticket if you still have issues. Also please use the bug report template when creating issues, it help us a lot to have some context (which device you are using, etc...)

closed 2023-09-08 by jimmyzumthurm