polarofficial/polar-ble-sdk #354: Polar H10 - Android - PolarBleApi callback functions are not getting called

teunklijn, 2023-03-28:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
I'm connecting to the Polar H10 on a OnePlus 8T with Android 13. The connection proces completes correctly, but only some of the `PolarBleApi` callback functions are ever called when we auto connect. `disInformationReceived`, `batteryLevelReceived`, `polarFtpFeatureReady` and `hrNotificationReceived` are never called it seems like. We are initializing the api with the required services. We see the same behavior when we run the example app. 

**How to Reproduce**
1. Run the latest example app
2. Connect to a Polar H10
3. Check the logs to see if the `disInformationReceived` or `batteryLevelReceived` callbacks are called.

**Expected behavior**
The logs show "DIS INFO uuid..." and "BATTERY LEVEL:...."

JOikarinen, 2023-03-29:
Hi @teunklijn 

There are known problems with Android 13 and H10. 

Here is the official statement related to H9 and H10 and Android13 compatibility issues:

"We are sorry to hear about this trouble with your heart rate sensor. The issue seems to be caused by the recent Android 13 update, and we are aware that the problem has been fixed already to certain phones like Samsung and Google Pixel, so please check first if there's a newer Android version available for your phone. It might solve the issue you’re experiencing.
Despite of this, our team at Polar has been working on a solution for this issue as well to ensure all our customers get their sensors back to work with the Android 13, and therefore we will release a new firmware version for our heart rate sensors. The date is yet to be confirmed, but you will be notified about the update availability via Polar Flow and Beat apps and at the Updates at support.polar.com."

JOikarinen, 2023-03-29:
Also reported in #350

closed 2023-03-29 by JOikarinen

aten040791, 2024-09-01:
Why so many bugs