polarofficial/polar-ble-sdk #355: Offline accelerometer recording in SDK mode only works for few minutes

RealLast, 2023-03-29:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [X] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
I want to use the offline recording feature of the PolarBleSdk in order to record accelerometer data to the internal storage of a Polar Verity Sense. This works well in normal mode, where accelerometer data is recorded at 52Hz. I can start and stop recordings, download and delete them as expected. 

Now, in order to optimize battery life time and storage duration, I want to record data at 26Hz instead of 52Hz. For this, I wait for SDK_MODE_FEATURE and OFFLINE_RECORDING_FEATURE to become available. Afterwards, I enter the SDK mode using PolarBleApi.enableSDKMode() and then start the offline recording again. This also works, the Polar Verity Sense switches to SDK mode which I notice by the optical HR sensor turning off and the LED on the side blinking in red, green and blue. The recording also starts successfully, and I can also download it later. However, and here is the problem: This only seems to work for 10 to 20 minutes. Afterwards, the LED of the Polar Verity Sense stops blinking and it seems to be the device has turned off. If I press the button, it starts again normally. I can then download the previous recording and start a new recording.

**How to Reproduce**
Use the PolarBleApi to enable SDK mode on the Polar Verity Sense. Afterwards, start an offline recording at 26Hz using PolarBleApi.startOfflineRecording. Note: I am using the SDK in Java, not Kotlin. So I am calling the corresponding functions from java.

I pass the following PolarSensorSettings:
PolarSensorSetting.SettingType.SAMPLE_RATE = 26
PolarSensorSetting.SettingType.RESOLUTION  = 16
PolarSensorSetting.SettingType.RANGE = 8
PolarSensorSetting.SettingType.CHANNELS = 3

During initialization of the api (PolarBleApiDefaultImpl.defaultImplementation(...)), I activate the following features:
PolarBleApi.PolarBleSdkFeature.FEATURE_POLAR_SDK_MODE
PolarBleApi.PolarBleSdkFeature.FEATURE_BATTERY_INFO
PolarBleApi.PolarBleSdkFeature.FEATURE_POLAR_OFFLINE_RECORDING
PolarBleApi.PolarBleSdkFeature.FEATURE_POLAR_ONLINE_STREAMING
PolarBleApi.PolarBleSdkFeature.FEATURE_DEVICE_INFO

**Expected behavior**
I expect the Polar Verity Sense to stay activated and record in SDK mode until I stop the recording (or it runs out of battery).

Any help is highly appreciated, thank you very much!

Best regards

samulimaa, 2023-04-11:
Thank you for reporting this. I have created an internal bug ticket to investigate this further.

merschel, 2023-06-16:
Hi @jimmyzumthurm 

after further tests in normal mode I have to report that it does not work properly either. 

I wore the device and it was in normal mode and the device remained on, furthermore the green LEDs for PPG measurement always were always on when I checked. 

I started the offline recording with an empty hard disk. 

In 3 tests the offline recording stopped after 99 minutes, 72 minutes and 47 minutes. When you try to start it again, you get the Full Disk error message. Although clearly less than 16 MB of the memory are occupied. At least that is what the API reports. 

It could be related to this issue https://github.com/polarofficial/polar-ble-sdk/issues/379

jimmyzumthurm, 2024-03-07:
Closing this as fixed in latest release available to Flow App

closed 2024-03-07 by jimmyzumthurm