polarofficial/polar-ble-sdk #358: SDK on macOS

mpepito13, 2023-04-04:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ x] Device is not relevant for this question

**Description:**

Hello,

I would like to know why the SDK is mobile only ? 
Is there any plan to port it to macOS ?

Thanks

samulimaa, 2023-04-11:
Hello, unfortunately, there are currently no plans to port SDK to new platforms such as macOS.

jimmyzumthurm, 2023-09-08:
Hi @mpepito13 , 
It is difficult to judge. There are people that already did porting to other platforms and languages (), but having no experience in MacOS application development I cannot say. We do not have the resources to support the SDK on other platforms than Android and iOS for now, so unfortunately support with remain on those 2 platforms only for now.

closed 2023-09-08 by jimmyzumthurm