polarofficial/polar-ble-sdk #359: Calling `getAvailableOnlineStreamDataTypes` throws `PolarBleSdk.PolarErrors.notificationNotEnabled` with the H9

Rexios80, 2023-04-07:
**Platform on which you observed the bug:** 
- [x] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [x] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Getting `PolarBleSdk.PolarErrors.notificationNotEnabled` when calling `getAvailableOnlineStreamDataTypes`

This is confirmed working for the H10 and Verity

**How to Reproduce**
Call `getAvailableOnlineStreamDataTypes` for a Polar H9

**Expected behavior**
It works like with the H10 and Verity

**Screenshots and logs**
[image]

**Additional Information**
Not sure how useful this is, but here's the code for the plugin: https://github.com/Rexios80/polar
The example app in that repo can reproduce the issue

samulimaa, 2023-04-19:
Hi, `getAvailableOnlineStreamDataTypes` is part of the Polar Online Streaming API which is only supported by Verity Sense, H10 and OH1 devices. But I think that the thrown error could be improved to reveal the actual cause.

Rexios80, 2023-04-19:
That makes sense, except heart rate is part of streaming now which is why I expect that call to not fail on any device that supports heart rate streaming. I would expect `getAvailableOnlineStreamDataTypes` to return heart rate on every Polar device. I suppose I can just assume that heart rate is available, but it's still weird.

Rexios80, 2023-04-19:
Also, `sdkFeatureReady` is returning `onlineStreaming` when connected to the H9. I'm waiting for that to even make the `getAvailableOnlineStreamDataTypes` call

[image]

Rexios80, 2023-07-04:
@samulimaa Is a fix for this planned?

Rexios80, 2023-09-11:
@samulimaa This seems like it would be trivial to fix. Should we except a solution for this soon or work around it?

Rexios80, 2025-04-21:
@JOikarinen @palmqvisti @TeemuLauPolar @andYgnite @jimmyzumthurm @pth-pe-gh 

This was reported 2 years ago and has had no response since. I now have a user of my plugin reporting this issue.

Possible solutions:
- Remove `onlineStreaming` from `sdkFeatureReady`
- Fix `getAvailableOnlineStreamDataTypes`
- Close this issue and I'll accept you don't want to fix it

rkangast, 2025-04-24:
Hello @Rexios80 

We are sorry that you had to wait for a response for too long. We are now investigating this issue. A probable solution is to move the checking of hr service availability in getAvailableOnlineStreamDataTypes method away from  blePMDClient.clientReady rxjava chain. I cannot promise that this fix will be in our next SDK release, but most likely it will be in the next subsequent release.

rkangast, 2025-05-09:
Hello @Rexios80 

Some interim information on this issue. I ended up doing a new API method "getAvailableHRServiceDataTypes" to solve this problem. It returns similar set of data as getAvailableOnlineStreamDataTypes method, so it should be straightforward to use instead. The getAvailableOnlineStreamDataTypes method could not be modified without bringing potential problems to other customers. The new API will be released in the next SDK release.

Rexios80, 2025-05-09:
@rkangast Did you also remove "onlineStreaming" from the list of available data types form the H9?

rkangast, 2025-05-12:
@Rexios80 

Yes I did, there is only "HR" as H9 does not support our PMD service that the other online measurement types use. But you can use H9 to measure online HR by calling startHrStreaming from PolarOnlineStreaming API. That API is using Bluetooth heart rate service the measure HR (for all our devices supported by SDK).

rkangast, 2025-05-16:
Hello @Rexios80 

polar-ble-sdk version 6.3.0 has been released and it has the [new API](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-android/com/polar/sdk/api/PolarOnlineStreamingApi.html#getAvailableHRServiceDataTypes(java.lang.String)) in place.