polarofficial/polar-ble-sdk #364: Cannot connect to device after connecting in system bluetooth settings

Rexios80, 2023-05-02:
**Platform on which you observed the bug:** 
- [x] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Connecting to a Polar device in the system bluetooth settings makes it to the SDK cannot connect to it

**How to Reproduce**
- Connect to Polar device in system bluetooth settings
- Try to connect with SDK

**Expected behavior**
Device connects like normal

**Screenshots and logs**
I haven't been able to reproduce this on my devices, but I have had several users able to reproduce this consistently. Is this intended/known behavior or a bug?

chrisgblr, 2023-05-16:
I have the same issue.
There is an setting in the polar flow app to enable 2 bluetooth devices. But by default it is turned off.
If that setting is enabled, you can connect via the system bluetooth settings and the SDK simultaneously.
Wouldn't it make sense to enable that setting by default to prevent connection issues?

andYgnite, 2023-07-10:
Hi @chrisgblr,
thank you for your suggestion. 

Although it may make sense to have dual connection enabled by default for certain use cases, I believe it is important for end users to be aware of this setting.

jimmyzumthurm, 2023-09-08:
@chrisgblr It is intended that dual-connection mode is not enabled by default and there is no plan to change that

closed 2023-09-08 by jimmyzumthurm