polarofficial/polar-ble-sdk #368: Request for Assistance in Porting PolarH10 ECG Data Reading Code for Linux

gd1925, 2023-05-12:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
My objective is to develop a ROS node that is capable of reading Heart Rate and Heart Rate Variability data from PolarH10 device. As it stands, the existing codebase is designed to function of app-based platforms, and  I am keen on getting this code to operate on a Linux OS environment, or Windows, to better fit the scope of my project.

I understand from various issues here that the out-of-the-box support for the adaptation might not be readily available. However, I am curious if someone within our community has successfully ported this code to run on Linux or Windows systems. 

In case a direct port is not possible, I would greatly appreciate any pointers or guidance that you can provide to help me. Any insights into the possible hurdles, or suggestions on alternate ways to would be highly beneficial. Thank you in advance for your time and consideration

erikboto, 2023-05-26:
Do you want to read the ECG data stream in order to calculate the RR-intervals yourself? If so you need to use the custom Polar services, which is possible to do using the information here but not super straight forwards. If you are just interested in the heart rate and the RR-intervals as detected by the Polar H10's internal processing, you can just read the data from the H10 using the standard [Heart Rate Profile](https://www.bluetooth.com/specifications/specs/heart-rate-profile-1-0/).

stuartlynne, 2023-06-13:
Polar H10 supports the standard GATT Heart Rate Service if you only need heart rate and RR interval information.

There are numerous open-source projects that show how to implement GATT Heart Rate access in various languages. 

Here is a good starting point for Python https://github.com/zacharyedwardbull/pycycling

Supports access to:

- Battery Service (BAS)
- Cycling Power Service (CPS)
- Cycling Speed and Cadence Service (CSCS)
- Elite Sterzo Steering Service (STERZO)
- FiTness Machine Service (FTMS)
- Heart Rate Service (HRS)
- Rear View Radar (RDR)
- Tacx Trainer Control (ANT+ FE-C over BLE)

Pittconnect, 2023-07-25:
@gd1925 Hello there I am curious to learn if you were successful in extracting HR and HRV data along with Timestamp from PolarH10. If you have please share any pointers or challenges you may have face in doing such an extraction.

stuartlynne, 2023-07-25:
See here for a test script to access Polar H10 from Python. Shows how to get ECG and ACC data. Tested in Linux and Windows.

https://github.com/stuartlynne/pypolartest

gd1925, 2023-07-26:
Hi all, 
Thank you very much for all the replies and the links. Really appreciate. 
@Pittconnect , yes I managed to some results using these two repositories - https://github.com/kbre93/dont-hold-your-breath 
and https://github.com/fg1/BLEHeartRateLogger/blob/master/BLEHeartRateLogger.py
For HRV computations, I am also using standard deviation and the rmssd values from the IBI values. I am using the general python code for the computation.

closed 2023-10-12 by samulimaa