polarofficial/polar-ble-sdk #370: Getting a different address in DeviceInfo when deviceConnected is called

aiartsev, 2023-05-22:
***Platform your question concerns:*** 
- [ ] Android 
- [ x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
When the deviceConnected callback is called in the `PolarBleApiObserver`, we sometimes get a different address in the `PolarDeviceInfo` from the same device. Is this expected, or could it be an indication of some issue?

jimmyzumthurm, 2023-09-08:
Hi,
Polar H9 and H10 are using Random Static Address instead of Public Device Address, meaning that the address will change every time the battery is removed/replaced and after a firmware upgrade. Let me know if that answer your issue.

closed 2025-07-03 by rkangast

mattisipola, 2025-07-03:
Since 2024-05-27 H9 v1.2.1 and H10 v3.3.1 have been using static Public Device Address. It will not change anymore.