polarofficial/polar-ble-sdk #371: Is there any difference between R-R interval and IBI?

gd1925, 2023-05-24:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Are IBI and R-R intervals the same? There are some sources which report them to be the same but some dont. Could someone please clarify?

lukesday, 2023-05-24:
RR is the time elapsed between two successive R-Waves and is used with ECG recording. IBI is the interbeat interval or interval between heart beats calculated from PPG data.

The main difference here is that RR comes from the electrical signal measured with an ECG and IBI is from the blood flow measured with PPG using a light based sensor. These values are usually interchangeable but IBI is subject to more forms of interference such as movement or ambient light and as such can be less accurate. 

https://www.analog.com/en/technical-articles/examining-interbeat-interval-measurement-capability-of-wearable-healthmonitoring-algorithms.html 
https://www.sciencedirect.com/topics/nursing-and-health-professions/rr-interval

gd1925, 2023-05-29:
Hi @lukesday, Thank you very much for your reply and the explanation. 
But since PolarH10 does not have PPG, can the RR values given by it be used as IBI? Can these R-R intervals be further used to get the HRV values?
Also if a device can detect give both PPG and ECG (so we have both IBI and R-R Intervals), which of the two measurements can be used to more reliably get the HRV values? Thank you once again!

gd1925, 2023-06-01:
Hi, I realise you have answered my questions in your message itself. Sorry about it.  Actually my question is more like - since PPG is not availiable using Polarh10 device, can it be assumed that wherever IBI is mentioned (for PolarH10) we can assume it to be R-R intervals?

lukesday, 2023-06-01:
For the purposes of calculating HRV, R-R intervals and IBI can be used interchangeably.

"Beat-to-beat data needs to be collected in order to [properly calculate HRV](https://www.labfront.com/course-video/time-domain-hrv). This data has many names attributed to it including interbeat interval (IBI), beat-to-beat interval (BBI), heart period, pulse interval, and RR interval.:

labfront.com/article/interbeat-interval

jimmyzumthurm, 2023-09-08:
> Actually my question is more like - since PPG is not available using Polarh10 device, can it be assumed that wherever IBI is mentioned (for PolarH10) we can assume it to be R-R intervals?

Hi @gd1925 , @lukesday already provided complete and accurate answers to your questions, but just to clarify : 
In our ecosystem, the Beat-to-beat data produced by ECG based sensor (H9, H10), correct terminology is RR interval,
and the data produced by PPG based sensor (OH1, Verity Sense) is PP interval (or PPI).

You will find such terminology already in Polar BLE SDK : for example RR interval from H10 can be access from `PolarHrData `class : 
https://github.com/polarofficial/polar-ble-sdk/blob/fdb5bd0c485eecbfc857708f8f9c65efd02052ae/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarHrData.kt#L15C16-L15C29

closed 2023-09-08 by jimmyzumthurm