polarofficial/polar-ble-sdk #374: On SDK Mode listOfflineRecordings does not work

merschel, 2023-06-08:
**Platform on which you observed the bug:** 
- [ ] Android 
- [ x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**

Hi,

On SDK mode the function listOfflineRecordings response always in responseError(errorCode: 202).

In the normal mode listOfflineRecordings works fine.

jimmyzumthurm, 2023-06-13:
Hi @merschel  

Error 202 is telling me that you are trying to read or list the files from one of those 3 states : 

1) Device was booted in exercise mode (also called internal recording). You would need to reboot your device in sensor mode (the "heart")
2) Device was booted in swim mode. Same as 1)
3) SDK mode is enabled. Currently you need to disable SDK mode to be able to get your files. 

We will add documentation to explain better these limitations. Note that the first 2 will probably remain like that, but we will discuss internally if we could allow syncing of files from SDK mode.

jimmyzumthurm, 2023-06-13:
@merschel I agree with you that is not intuitive and also missing from documentation, that will be added. These is historical reason for that block, which may not be valid anymore since we introduced the offline recording feature.

jimmyzumthurm, 2023-09-08:
@merschel FYI, in next release (can't give any ETA yet), it will be possible to fetch recordings from SDK mode. 

Also note that there is similar "SYSTEM BUSY" block when device is in exercise mode / swim mode. Unfortunately these blocks will need to remain, because this is needed to prevent sync with Flow service during exercise

closed 2023-09-08 by jimmyzumthurm